stlehmann/pyads #191: not working on raspian?

zuensler, 2021-02-09:
Does pyads work on a raspi with raspian?

I'm able to create the route:
```python
import pyads
SENDER_AMS = '1.2.3.4.1.1' #ID of my raspi
PLC_IP = '10.100.1.235' #IP of the PLC
PLC_USERNAME = 'Administrator'
PLC_PASSWORD = '****'
ROUTE_NAME = 'ScannerRoute'
HOSTNAME = '10.100.1.244'  # Ip of my raspi
PLC_AMS_ID = '10.100.1.235.2.1' #ID of my IPC
pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, PLC_USERNAME, PLC_PASSWORD, route_name=ROUTE_NAME)
```

the route appears on my IPC

oke now I want to read a value from my PLC

```python
import pyads
netID = "10.100.1.235.1.1"
plc = pyads.Connection(netID, 851)
plc.open()
i = plc.read_by_name("MAIN.test_val", pyads.PLCTYPE_INT)
print(i)
```
and I get a timeout:
```
root@scanbox-input:/home/scanbox# python3 scanner.py
2021-02-09T13:43:15+0000 Info: Connected to 10.100.1.235
Traceback (most recent call last):
  File "scanner.py", line 12, in <module>
    i = plc.read_by_name("MAIN.test_val", pyads.PLCTYPE_INT)
  File "/usr/local/lib/python3.7/dist-packages/pyads/ads.py", line 832, in read_by_name
    check_length=check_length,
  File "/usr/local/lib/python3.7/dist-packages/pyads/pyads_ex.py", line 1028, in adsSyncReadByNameEx
    handle = adsGetHandle(port, address, data_name)
  File "/usr/local/lib/python3.7/dist-packages/pyads/pyads_ex.py", line 829, in adsGetHandle
    PLCTYPE_STRING,
  File "/usr/local/lib/python3.7/dist-packages/pyads/pyads_ex.py", line 690, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861).
2021-02-09T13:43:20+0000 Info: connection closed by remote
```

works fine with Windows...

Thanks!

stlehmann, 2021-02-10:
@zuensler pyads does not support Raspbian. Mainly because we are not able to test on this architecture.

zuensler, 2021-02-10:
ok, thanks for the info!

closed 2021-02-10 by zuensler

stlehmann, 2021-02-12:
@zuensler I just got the info that pyads works well on Raspbian. So it should work out for you too. Most problems arise because of wrong routing setttings. Maybe you should give them a second look.