stlehmann/pyads #196: Is read the docs updated correctly?

chrisbeardy, 2021-02-13:
I saw you made a new release yesterday which updated all the docs, however looking at readthedocs, they do not seem to be displaying correctly?

[image]

closed 2021-02-14 by chrisbeardy

chrisbeardy, 2021-02-14:
nice work on the docs btw.

stlehmann, 2021-02-14:
@chrisbeardy thanks. Looks like RTD is still displaying the old version for you. For me it loks different:

[image]

Did you reload the page?

chrisbeardy, 2021-02-14:
This sorted itself out after you pushed some more commits today.