stlehmann/pyads #201: Simplify Type Handling

koppa, 2021-02-17:
Currently are there are multiple different ways how to pass a type to the pyads module:

* As Strings:
   * "LREAL"
   * "ARRAY[0..10] of LREAL"
* pyads.PLCTYPE_X
* ctypes type (Structures, etc...)
* "string" for the struct module

I would propose to unify this handling, that every method accepts the same typedefintions.
For example the AdsSymbol has implemented its own parsing code.

RobertoRoos, 2021-02-17:
I agree. We could pool all type interpretations into a static function that will either produce a ctype-like type, read to be used by Ads or raises an exception.
This function can then be used by each read and write function, and by AdsSymbol.

chrisbeardy, 2021-02-17:
As far as the public API, I see 3 types ways of defining types available?:

- For read/write_by_name,, read/write_structure_by_name, using pyads.PLCTYPE_*
- For AdsSymbol, using  a string, "BOOL"
- Notification callbacks using ctypes.Structures
 
@koppa  What is the "string" for the struct module?

I can see why each method was used to acheive its desired functionality. But I can also need a need to unify this.

## Unifying the type info for the user

I would argue that the ctypes.Structure could be left as is, it seems quite a standalone use case? and possibly superseeded by the structure def approach which we could point new users too?:

[image]

The original way from read/wirte_by_name was using pyads.PLCTYPE* and then if you needed an array, multiplying it by array size:

[image]

however since the introductions of read/write_list_by_name and the ADSSymbol pyads seems to be moving toward a more finding the type and caching the symbol address etc, which makes a lot of sense and abstracts a lot of the detail from the user.  For unifying the ADS_SYMBOL and others, I see currently not too great a harm leaving it as it right now for two reasons.

1) It will bring braking changes, not that that is a big issue, but I think right *now* its not a big enough problem to refactor it.
2) I *believe* for a majority of use cases going forward (especially now #195 has been merged too) one will not provide plc type info (except for write/read_structure_by_name), as pyads will get the info itself , so I see it more of an advanced feature for speeding up your code or for edge cases, perhaps such as #189.

The deveolpment of pyads should be mindful not to keep adding more type definition ways and if it were to unify in the future, in my opinion I would say when the user has/wants to provide the PLC type to pyads they should do so through using pyads.PLC_TYPE* and not string, as that provides a set of variables for the user to select , instead of writing the info as string. But thats just my preference, I don't have any hard justifications why.

RobertoRoos, 2021-02-17:
- About the breaking changes, I'm not sure whether that's the case. The idea is to make all function more flexible. Meaning that they will accept different types of type-definitions (need to work out the jargon here :p). We can keep an eye on backwards compatibility.
- I agree that types written out by users should be of the `constants.PLCTYPE_*` sort. There is no real advantage in letting users write out other methods. However, different types of types (ouch) need to be handled internally, for example because `getSymbolInfo` will return a string ADS-type.

chrisbeardy, 2021-02-17:
> 
> 
>     * About the breaking changes, I'm not sure whether that's the case. The idea is to make all function more flexible. Meaning that they will accept different types of type-definitions (need to work out the jargon here :p). We can keep an eye on backwards compatibility.
> 

I was thinking more from a user perspective if type input to say read_by_name was to change to a string, this would be a breaking change. If the intend would be to not change this but just change type handling internaly then yes, no breaking change needs to happen.

stlehmann, 2021-02-17:
I agree with @RobertoRoos that types written by the user should use constants.PLCTYPE_. As @chrisbeardy stated this has the advantage that all basic types are defined in our constants module which makes it easier for the user to select them. Structures and lists are also defined by using the PLCTYPE constants so that is fitting well in my opinion.

How about changing AdsSymbol API in a way that the user needs to provide PLCTYPE_ or structure defnition instead of strings if he wants to define the type manually. This way the AdsSymbol API would become more compatible with the read/write functions and would also allow access to structures and lists. And as stated above for internal use the STRING datatype can still be used.

RobertoRoos, 2021-02-17:
Sounds good.

If you'll allow me to be petty for a moment: that is how I made it in the first place, with the `symbol_type` field being as flexible as possible.

I can make an update on Friday. I'll see if someone beats me to it.

stlehmann, 2021-02-17:
> If you'll allow me to be petty for a moment: that is how I made it in the first place, with the symbol_type field being as flexible as possible.

Yes, I was also thinking about our little conversation on this matter, BIG SORRY I didn't have the whole situation in mind at that time. 🙏

stlehmann, 2021-02-17:
> Shall we remove support for a string type in explicit Symbol creation then? To me there is no real point in removing it. More flexibility is more better. In our documentation and quick-start we can point the user to the correct method (being the constants)

I personally like it more if there is only one way to do things. But on the other hand I see no real harm if one can define a type also by using a string, plus we won't break the current API if we still allow it so I would keep string-support. But I agree that we should encourage the users to use constants here.

stlehmann, 2021-03-18:
I think we kind of solved this issue with #214 so I will close it for now. If anyone has some more ideas feel free to add further comments or reopen.

closed 2021-03-18 by stlehmann