stlehmann/pyads #204: make StructureDef type available as import?

chrisbeardy, 2021-02-17:
The custome type def StructureDef:

```python
StructureDef = Tuple[
    Union[Tuple[str, Type, int], Tuple[str, Type, int, Optional[int]]], ...
]
```

Would this make sense to add it so that users can import it and do this?

```python
from pyads import StructureDef

COMMAND_STRUCTURE: StructureDef = (
     ("actionID", pyads.PLCTYPE_UINT, 1),
     ("commandID", pyads.PLCTYPE_UINT, 1),  # Change to trigger command input
)

def _amend_def_var(var, original_def: StructureDef) -> StructureDef:
    return tuple(((var + "." + i[0]),) + tuple(i[1:]) for i in original_def)
```

stlehmann, 2021-02-17:
That's nice. We can do that.

stlehmann, 2021-02-18:
@chrisbeardy thanks for taking care of it

closed 2021-02-18 by stlehmann