stlehmann/pyads #207: AdsSymbol: add full support for structures

stlehmann, 2021-02-17:
At the moment AdsSymbols don't support structures. 

The nicest way I can think of would be to read a structure definition via some ADS command? But I doubt that there exists such a command. 

So maybe we can use the approach of providing a structure definition as in read_structure_by_name? @RobertoRoos what are your thoughts on this?

RobertoRoos, 2021-02-17:
Yeah from what I've seen in my tests this afternoon, you cannot get the complete struct definition over ADS. So the user will have to provide it.

Users could simply provide it with the `symbol_type` argument. We can capture a Tuple type (which should never be passed besides structs) and set an internal flag `_is_struct = True`.

In read()/write() we can add an if-else to use the `read_structure_by_name()` in case `_is_struct == True`.

We can keep the returned value identical to what's returned by `read_structure_by_name()`, which is an `OrderedDict` I think.

stlehmann, 2021-02-17:
I like the idea to include read_structure_by_name. That keeps the codebase small.

> symbol.value[field] = 3.14

Oh that's tricky. I think in this case the wrapper would return an object and __setitem__ would be called. This mean the getter needs to return a wrapper object that uses __setitem__ to update the target value. I don't think we should support that. Maybe we can handle structs more the "tuple-way":

```python
   struct = symbol.value
   struct["a"] = 123
   symbol.value = struct
```

closed 2021-03-07 by stlehmann