stlehmann/pyads #210: parameter undefined

asdomar45, 2021-02-18:
Hi, i'm working on university thesis about ADS connections. I read that:
>>> plc.read_by_name('GVL.sample_string', pyads.PLCTYPE_STRING)
'Hello World'
>>> plc.write_by_name('GVL.sample_string', 'abc', pyads.PLCTYPE_STRING)
>>> plc.read_by_name(adr, 'GVL.sample_string', pyads.PLCTYPE_STRING)
'abc'

but when i use the command plc.read_by_name i don't know what the adr parameter is.
If i use the plc.write_by_name('global.sample_string', 'Hello World!', pyads.PLCTYPE_STRING),  command and then the plc.read_by_name('global.sample_string', pyads.PLCTYPE_STRING) it doesn't print 'Hello World!' as i expected.
Do you know what the problem may be?
Thanks

stlehmann, 2021-02-18:
That is actually a bug in the docs. It should be this:

```python
plc.read_by_name('GVL.sample_string', pyads.PLCTYPE_STRING)
```

stlehmann, 2021-02-18:
Fixed by 463ca3228f74a99e84972c8aac8062ba8f41456c.

closed 2021-02-18 by stlehmann