stlehmann/pyads #211: Use GitHub actions for CI

pylipp, 2021-02-18:
Are there any objections against using GitHub actions for CI? It's faster than travis (which queues test suites for quite some time) and a native part of the code hosting platform. Also travis changed it's pricing model last fall, so OS projects have less free build minutes.

With GitHub actions it's straight forward to run both Python tests, and even publish a package automatically to PyPI (not sure if @stlehmann currently manually publishes them?). The config for a side project of mine is [here](https://github.com/pylipp/financeager/tree/master/.github/workflows). If desired I'll copy and adjust it for pyads.

stlehmann, 2021-02-18:
> Are there any objections against using GitHub actions for CI?

Actually my only objection is that I know Travis CI well and have been working with it for a long while now. I haven't used github actions so far. 

> With GitHub actions it's straight forward to run both Python tests, and even publish a package automatically to PyPI (not sure if @stlehmann currently manually publishes them?)

Automatic publishing on PyPi sounds like a nice feature. Indeed I do the upload manually since the beginning of time.

> The config for a side project of mine is here. If desired I'll copy and adjust it for pyads.

If you like you can make a PR to setup CI on Github Actions. This way we can run both CIs in parallel and eventually change to Github if it performs better.

stlehmann, 2021-02-24:
@pylipp Thanks for the bump towards Github Actions. I introduced them with #221. Also Upload to PyPi should happen automatically on release.

PR #221 (2021-02-24): Use Github Actions for CI 
This fixes #211
closed 2021-02-24 by stlehmann