stlehmann/pyads #224: AdsSymbol returns Bytes not STRING

chrisbeardy, 2021-03-02:
I'm trying to use ADSSymbol to read/write a string but I'm getting TypeError if I pass a string to write and when i read I get a list of bytes and not the string representation. Is this intended behaviour?

PLC declaration:
[image]

[image]

[image]

chrisbeardy, 2021-03-02:
doing this in my end code sorts it our for me but just wondering if this is expected?

```python
self._display_plc_symbol_type = self._display_plc_symbol.symbol_type
if "STRING" in self._set_plc_symbol_type:
    value = value.encode()
self._set_plc_symbol.write(value)

value = self._display_plc_symbol.read()
if "STRING" in self._display_plc_symbol_type:
    value = value.decode()
```

chrisbeardy, 2021-03-03:
UPDATE: this also seems to affect PR #195 trying to read a string using `read_by_name` and not pasisng in datatype now also reads a bytes list and not a string.

[image]

chrisbeardy, 2021-03-03:
it appears to recognise string stypes as PyCArrayTypes when using adsGetSymbolInfo thereforedoes not do the encoding.

[image]

Should  pyads.ex be modified, There are a few places that **may** need modification. `adsSyncReadReqEx2`, `adsSyncWriteReqEx` and `adsSyncReadWriteReq2" seem to be the main ones but there are loads of places where a check is made against PLCTYPE_STRING to determine what to do with the response and encode/decode the bytes. 

e.g.
[image]

Not sure wht the knock on effects to AdsSymbol would be (or anything else) here if we modified that a PyCArrayType is also encoded/decoded, or should this be done further up the tree in the callin functions, e.g. read_by_name

not sure what @RobertoRoos thinks here?

chrisbeardy, 2021-03-04:
Just did a further test, it appears version 3.3.2 and 3.3.3 using `var = plc.get_symbol()` then `var.read()` on a string type is giving me a NoneType Error

[image]

RobertoRoos, 2021-03-04:
The difference is the following:

```
plc = pyads.Connection('127.0.0.1.1.1', pyads.PORT_TC3PLC1)

with plc:
    value1 = plc.read_by_name('GVL.my_text')
    print(value1)  # Shows bytearray

    value2 = plc.read_by_name('GVL.my_text', pyads.PLCTYPE_STRING)
    print(value2)  # Shows proper string

    value3 = plc.read_by_name('GVL.my_text', pyads.PLCTYPE_STRING * 50)
    print(value3)  # Shows bytearray
```

SymbolGetInfo will neatly return "STRING(80)", but it looks like the internal ads functions will return a string if _no_ size is given.

This can be solved by changing the `get_type_from_str` method to ignore the string length.

@stlehmann do you know if there was deliberate reasoning here? Maybe we should modify the internal functions to return a string for both 'sized' and 'unsized' strings.

RobertoRoos, 2021-03-04:
By adding the following to `get_type_from_str()` both `AdsSymbol` and `read_by_name` will return a proper `str` as expected:

```
            scalar_type = AdsSymbol.get_type_from_str(scalar_type_str)

            if scalar_type == constants.PLCTYPE_STRING:
                return scalar_type  # Do not include the length - This way the ads
                # functions will return a python string, instead of a byte array

            if scalar_type:
                return scalar_type * size
```

RobertoRoos, 2021-03-04:
I'd suggest we approach this from `pyads_ex`. See new PR.

Analysing the passed ctypes classes is unfortunately very awkward... Getting the type of a ctypes array seems difficult.

closed 2021-03-06 by stlehmann