stlehmann/pyads #227: Change default port in documentation from PORT_SPS1(==PORT_TC2PLC1) to PORT_TC3PLC1

mtroback, 2021-03-08:
As a beginner I was a bit confused about which port to use. Once I figure out I should use the PORT_TC3PLC1 everything worked.

I can only assume that anyone new to this project is using TwinCAT3 and thus it makes sense to updated the documentation to reflect this. Maybe this will save some time for some other poor developer whom doesn't double check that the correct port is used when trying to connect. I even had to contact Beckhoff support to be sure that I was supposed to use the target AmsNetID to connect to my PLC.

I've prepared the changes at https://github.com/mtroback/pyads but this is the first time I'm doing a pull request so I will try to follow the guidelines. Still need to run black for syntax and tux for testing. And I guess this issue is also required for reference when creating the pull request. Although I'm not sure if I can make the changes on master or if it is necessary to create a branch on my fork.

stlehmann, 2021-03-18:
@mtroback thanks for the feedback. That is actually a good idea. I will fix this asap. If you like you can also drop a PR for it.

stlehmann, 2021-04-22:
Fixed with 9ac00bd0e96d2b570d5f6a0ced578c994ffcb59a

closed 2021-04-22 by stlehmann