stlehmann/pyads #229: Get structure from the PLC

jpulte, 2021-03-12:
Hi. I am new here and i am also new with pyads. To work with pyads with structures you have to define the structure beforehand, that have to match with the structure defined in the PLC. Is there a way to get the content of the structure (name, datatype, value) from the PLC without defining it before.

chrisbeardy, 2021-03-16:
Hi, there is no direct and easy way of getting the structure beforehand unfortunately. However there is the command `get_all_symbols` . This will display all the symbol info but will show one symbol per each item in the structure. Knowing the variable  name that is of the type of your structure you could work out the names and types of what is on the structure and build the definition of the structure dynamically.

chrisbeardy, 2021-03-16:
E.g. if I have a structure of plc Def:

```
TYPE test :
STRUCT

a: BOOL:
b: INT;

END_STRUCT
END_TYPE
```

Then I declare :

var : test:

If I do get all symbols you get for a symbols the following. 

Symbol 1: name : MAIN.var.a, type: BOOL
Symbol 2: name : MAIN.var.b, TYPE: INT

So you could scrape through and figure it  out from that.

stlehmann, 2021-03-18:
@jpulte thanks for passing by. I kindly want to bump you on using stackoverflow.com for asking questions on pyads usage. Github issues should be used for reporting bugs or enhancement proposals. Thanks :)

closed 2021-03-18 by stlehmann