stlehmann/pyads #236: Symbol device notifications in test server

RobertoRoos, 2021-04-09:
I think that device notifications are not called from the test server. Could that be right?

In `tests/test_connection_class.py` there is `test_decorated_device_notification()`, which sets up a callback for a notification and the value is written to. But the print inside the callback doesn't show.

Anyone remember this having ever worked?

It could be nice to have it as a test inside pyads itself, but I also need it in a test of another application I am making.

RobertoRoos, 2021-04-09:
I don't it does/did.

So adding a device notification ends with Python calling the `AdsSyncAddDeviceNotificationReqEx` function ([beckhoff docs](https://infosys.beckhoff.com/english.php?content=../content/1033/tcadsdll2/html/tcadsdll_funcadssyncadddevicenotificationreqex.htm&id=))

How can I find out which bytes are sent over by this function, and how to unpack them?

We can already do this for reading and writing data from the testserver. Where did you learn how to do that?

RobertoRoos, 2021-04-12:
I've found this describing the request and response for adding a notification: https://infosys.beckhoff.com/english.php?content=../content/1033/tcadsamsspec/html/TcAdsAmsSpec_ADSCmd_Read.htm&id=

But this doesn't involve the callback itself. The callbacks are wrapped in c-functions and stored in the global `pyads_ex.callback_store`. The callables are index by the AmsAddr and the notification handle: `Dict[Tuple[SAmsAddr, int], Callable]`.

So in the testserver I can keep track of which variables had notifications and refer to that list.
But how do I get the current `SAmsAddr` struct inside the handler? @stlehmann any ideas?

stlehmann, 2021-04-20:
@RobertoRoos sorry for the late answer. I need to look into this and will come back to you asap.

stlehmann, 2021-04-22:
Yes you are right. The testserver does not support device notification handling. All tests are just looking if DeviceNotification add/delete requests have been sent to the server.

stlehmann, 2021-04-22:
Because it is just a Testserver and not been meant to hold the full functionality of a true ADS server this was considered sufficient.

RobertoRoos, 2021-04-23:
Yeah, that makes sense for internal testing.
But having the complete cycle is useful when applications that use pyads require testing (like in my case the last few weeks).

stlehmann, 2021-04-23:
> Having the complete cycle is useful when applications that use pyads require testing

You are talking about automated tests of applications, right? So that is a different and more complex use-case. I see some benefit in letting developers test their applications with the testserver. But this means we will need to implement the full ADS functionality over time. Also there needs to be some documentation that explains which functionality is implemented properly and which is mocked or limited. I actually don't know if it's worth the effort?

RobertoRoos, 2021-04-23:
I see your point.
Though much of it is already implemented, it just needs some more documentation then. We could consider it an advanced feature, that doesn't need dummy-level documentation. We can just add some pages explaining the test server and add a table for the different handlers and their capabilities ('missing', 'mocked' or 'real').
So I would suggest we include this. But I guess I'm biased :p

chrisbeardy, 2021-04-23:
a "full" ads test server may be good as a seperate project, to which anyone can use in a CI pipeline for testing, would certainly be useful.

stlehmann, 2021-04-23:
> We could consider it an advanced feature ... We can just add some pages explaining the test server and add a table for the different handlers and their capabilities ('missing', 'mocked' or 'real').

That sounds good to me.

> But I guess I'm biased :p

Looks like it and that tells me you habe the drive to implement it 😄 

> a "full" ads test server may be good as a seperate project, to which anyone can use in a CI pipeline for testing, would certainly be **useful.**

I agree with @chrisbeardy but at the current state I consider the testserver too small to run as a separate project. We can keep that in mind while it grows.

PR #239 (2021-05-13): Add support for testserver notifications
Title says it all. Fixes #236.

It might be good to merge or abort #237 first, since I continued with the branch locally. We could yank out the refactoring again if it's unwanted.
closed 2021-05-13 by stlehmann