stlehmann/pyads #244: Publish new releases on conda-forge with Github Actions

stlehmann, 2021-05-14:
Github Actions work out fine so far. It would be great to automatically publish a release on conda-forge. However I couldn't find a fitting workflow so far. Maybe someone got an idea how to tackle it or some experience with this sort of thing. The conda-forge feedstock repository can be found here: https://github.com/conda-forge/pyads-feedstock

stlehmann, 2021-11-18:
Just came across this one. Considering that packaging on conda-forge has been a painful experience I ponder about droping conda-forge support alltogether and just maintain a single distribution channel (PyPI). Any thoughts are welcome.

chrisbeardy, 2021-11-18:
personally I see no need to add to the burden of maintenance by keeping a conda-forge package. However if someone wants to add it to GH actions and maintain it then great.

stlehmann, 2021-12-10:
That is my point of view as well. So I will drop support for conda-forge. Actually there should be no big action necessary at this point. I'll just remove the Anaconda badges from the Readme.

stlehmann, 2021-12-10:
Fixed with #296

closed 2021-12-10 by stlehmann

CagtayFabry, 2022-08-30:
first of, thank you for this amazing project making it so easy to work with ADS + Python 🚀 

I would really love to see `pyads` back on `conda-forge`.
If I understand correctly the main reason was the maintenance needed to keep supporting `conda-forge` releases next to the automated `PyPI` release pipeline?

I see that the old conda recipe is linked to the github source releases, maybe it would help to link the PyPI release instead and have that picked up by the conda-forge bot?
The build process on conda seemed to work from what I can see or is more configuration work needed on that front? 

Anyway, if you are interested to try setting this back up I am happy to help.