stlehmann/pyads #248: New testserver submodule missing in release 3.3.5

RobertoRoos, 2021-05-31:
I wanted to try out the new testserver in my application, but

```
from pyads.testserver import AdsTestServer
```

...fails, saying the `testserver` module does not exist.

The installed package contains the following files:

> .
> ├── __init__.py
> ├── ads.py
> ├── constants.py
> ├── errorcodes.py
> ├── filetimes.py
> ├── pyads_ex.py
> ├── structs.py
> ├── symbol.py
> └── utils.py

So it is missing the new testserver directory.

RobertoRoos, 2021-05-31:
I'm a dumbass, it hasn't been merged yet... I thought that with the version bump it had...

I wish Github had a commits tree view like Bitbucket does.

RobertoRoos, 2021-05-31:
Okay, I'm a double-dumbass. It was merged properly, the directory is right here: https://github.com/stlehmann/pyads/tree/3.3.5/pyads/testserver

So then there is a pypi problem. You can see the latest package is missing the new files: https://pypi.org/project/pyads/#files

RobertoRoos, 2021-05-31:
Tested this locally using `pip install .` and I can reproduce this problem that way.

I think I fixed it with adding the submodule in `setup.py`:
```
setup(
    # ....
    packages=["pyads", "pyads.testserver"],
```

But I've no clue whether this is conventional.

RobertoRoos, 2021-05-31:
The internet suggests it too: https://stackoverflow.com/questions/26528178/right-way-to-set-python-package-with-sub-packages

@stlehmann what do you think?

closed 2021-06-01 by stlehmann

stlehmann, 2021-06-01:
@RobertoRoos Thanks for making the PR. I merged it right away.

PR #249 (2021-06-01): Added pyads.testserver to setup.py
Subdirectory of the testserver was not included in pyads 3.3.5.
Fixes #248