stlehmann/pyads #254: Connection Raspberry Pi with TwinCat

FDuenne, 2021-06-29:
Hello,

I would like to establish a connection between pyads on a Raspberry Pi and a TwinCat on my local computer. 
With a python test script the connection works without any problems.

Unfortunately, the connection from the Raspberry Pi does not work.

I have followed these instructions:
https://pyads.readthedocs.io/en/latest/quickstart.html

How do I set the route and which IPs do I have to enter here? 

CLIENT_NETID = " "
CLIENT_IP = " "
TARGET_IP = " "

Thank you very much!

Translated with www.DeepL.com/Translator (free version)

chrisbeardy, 2021-07-08:
Hello, 

Please create a route using the following:

CLIENT_IP =  IP address of Raspberry Pi
CLIENT_NETID = IP adress of raspberry pi + ".1.1"
TARGET_IP =  IP address of the TwinCat computer. 

You can see that here:

[image]

chrisbeardy, 2021-07-08:
the target username and password should also be that of the TwinCat computer

simonschmeisser, 2021-07-08:
I'm currently also struggling with a similar issue. Could you please explain where I'd put the username and password?

FDuenne, 2021-07-08:
Hi guys,

I was able to solve the problem in a different way.

192.168.1.100

With " plc = pyads.Connection(pyads.ads.adsGetNetIdForPLC(' IP address of the TwinCat computer'), 851, ' IP address of the TwinCat computer') " I was able to establish the connection.

chrisbeardy, 2021-07-09:
The username and password will be the username and password of the target pc. In the case of a Beckhoff IPC the default is Administrator and '1'. If you are running it on a laptop or other device the username and password will be for that device. This is what you enter on the Ubuntu/pi side in the python script when adding the route.

chrisbeardy, 2021-07-09:
> Hi guys,
> 
> I was able to solve the problem in a different way.
> 
> With " plc = pyads.Connection(pyads.ads.adsGetNetIdForPLC(' IP address of the TwinCat computer'), 851, ' IP address of the TwinCat computer') " I was able to establish the connection.

Yes this would also work if you only know IP address and not and address. But you still need to have the route added first. Which only needs to be done once per device

chrisbeardy, 2021-07-10:
@FDuenne if you are able to connect please can you close this issue, many thanks.

FDuenne, 2021-07-10:
Yes sorry 👍

closed 2021-07-10 by FDuenne

Vishal99787, 2022-04-24:
Hello,

I am facing a similar problem. In my case, I am using real Beckhoff CX8190 PLC to communicate with Raspberry Pi 3 B+. Using Python script I am able to communicate with PLC but it doesn't work with Raspberry Pi. I have a question that do I have to add the AMS ID of Raspberry Pi(IP address .1.1) to the Router?
[image]
[image]

I am getting this error. If you have any ideas please guide me to solve my problem.

Vishal99787, 2022-04-26:
The route is added as you can see in the picture. However, I am getting the same error(Connection closed by remote(PLC) , Connection reset by peer) as I uploaded in the previous message.
Do we need to add the route details [which we have created] in the program of Raspi?

I could not find the actual root problem of the same.
Request you to guide & if possible, can we have a call?

[image]

Vishal99787, 2022-04-27:
Thanks you very @chrisbeardy for your help. I have managed to communicate between Beckhoff PLC and raspberry pi.

chrisbeardy, 2022-04-27:
What was your issue? Out of interest

Vishal99787, 2022-04-27:
I have changed the AMS ID of PLC. I make it IP address.1.1 of ethernet port where Raspberry pi is connected and it started working. Also , raspberry pi was not pinging PLC, when I connect it with internet.