stlehmann/pyads #255: pyads.pyads_ex.ADSError: ADSError: access denied (1827)

simonschmeisser, 2021-07-08:
I'm trying to connect to a TwinCat running on a Windows 7 PC via ADS from Ubuntu 18.04

I set up a route from the windows pc to the linux pc using these setting:
[image]

Now I run the following little piece of python:
```
>>> import pyads
>>> plc = pyads.Connection('10.2.0.216.1.1', 851)
>>> plc.open()
2021-07-08T18:53:22+0200 Info: Connected to 10.2.0.216
>>> i = plc.read_by_name("GVL.int_val")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 845, in read_by_name
    plc_datatype = self._query_plc_datatype_from_name(data_name, cache_symbol_info)
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 519, in _query_plc_datatype_from_name
    info = adsGetSymbolInfo(self._port, self._adr, data_name)
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 842, in adsGetSymbolInfo
    PLCTYPE_STRING,
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 681, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: access denied (1827). 
>>> i = plc.read_by_name("MAIN.i1_variables_Mikado._ui_i1_Mikado_ErrorCode")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 845, in read_by_name
    plc_datatype = self._query_plc_datatype_from_name(data_name, cache_symbol_info)
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 519, in _query_plc_datatype_from_name
    info = adsGetSymbolInfo(self._port, self._adr, data_name)
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 842, in adsGetSymbolInfo
    PLCTYPE_STRING,
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 681, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: access denied (1827). 
>>> i = plc.read_by_name("i1_variables_Mikado._ui_i1_Mikado_ErrorCode")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 845, in read_by_name
    plc_datatype = self._query_plc_datatype_from_name(data_name, cache_symbol_info)                                                                                                                                                                                           
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 519, in _query_plc_datatype_from_name                                                                                                                                                                
    info = adsGetSymbolInfo(self._port, self._adr, data_name)                                                                                                                                                                                                                 
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 842, in adsGetSymbolInfo                                                                                                                                                                        
    PLCTYPE_STRING,                                                                                                                                                                                                                                                           
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 681, in adsSyncReadWriteReqEx2                                                                                                                                                                  
    raise ADSError(err_code)                                                                                                                                                                                                                                                  
pyads.pyads_ex.ADSError: ADSError: access denied (1827).
```
Running the same code locally on the windows Pc will give me an error for the first read (because there is no such variable) and succeed for the third

Is there something like access control that I need to configure?

chrisbeardy, 2021-07-09:
Have you set the route up from Ubuntu to the windows pc?  In Linux you need to do that using python from the ubuntu pc as explained in the quickstart, the target is the windows pc in that scenario.  Using the add route dialog in Windows does not work with Linux unfortunately.

simonschmeisser, 2021-07-09:
When I try running `add_route_to_plc` I get:
```
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/ads.py", line 233, in add_route_to_plc
    added_net_id=added_net_id,
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 148, in wrapper
    return fn(*args, **kwargs)
  File "/home/sis/.local/lib/python3.6/site-packages/pyads/pyads_ex.py", line 294, in adsAddRouteToPLC
    raise RuntimeError(f"Unexpected response from PLC: {data!r}")
RuntimeError: Unexpected response from PLC: b"\x03f\x14q\x00\x00\x00\x00\x06\x00\x00\x80\n\x02\x00\xd8\x01\x01\x10'\x01\x00\x00\x00\x01\x00\x04\x00\x04\x07\x00\x00"

```

Am I supposed to use the windows credentials here? Or is that some Beckhoff default?

simonschmeisser, 2021-07-09:
I see you just answered that in #254 

However I don't understand why I cannot set the route from PLC (windows) to linux via the TwinCat route tool and the other direction is then automatically created like the comments in Quickstart suggest?

chrisbeardy, 2021-07-10:
The route does need to be set up on both sides yes. 

Usually on Windows to Windows, you can create both the target and the remote route at the same time from either the remote(client) or the target by checking the appropriate boxes on the right hand side of the dialog.

On Linux, the remote (client) route is set up automatically when you using the Connection class in pyads, however a target route still needs to be set up, the quick start suggests doing this by using the `add_route_to_plc()` function, this function is essentially placing an entry in the routes of the PLC/Windows PC as if you created it using the `Add Route Dialog` on the PLC/Windows PC. I believe you can still set the route up using the dialog as you did from Windows instead of using the `add_route_to_plc()` function, I believe if doing this the IP address, username and password would be that of Ubuntu.

Could you try setting up the Target route using the `add_route_to_plc()` function and see if this helps. 

I would first delete any routes you have created in the windows side then run:

```python
CLIENT_NETID = "10.2.0.75.1.1"   # net id of ubuntu
CLIENT_IP = "10.2.0.75"   # assuming  ip of ubuntu
TARGET_IP = "10.2.0.216"  # assuming ip of PLC/Windows PC
TARGET_USERNAME = the-windows-username
TARGET_PASSWORD = the-windows-password
ROUTE_NAME = "mikado"

# add a new route to the target plc

pyads.add_route_to_plc(
    CLIENT_NETID, CLIENT_IP, TARGET_IP, TARGET_USERNAME, TARGET_PASSWORD,
    route_name=ROUTE_NAME
)
```

Let me know how this works out.

simonschmeisser, 2021-07-10:
Unfortunately `add_route_to_plc` fails, no matter if I already have another route set up manually or not. Please see the Traceback two comments above

chrisbeardy, 2021-07-10:
OK, and if you create the route using the windows route dialog with Ubuntu's credentials you get the `ADSError: access denied (1827).`?

simonschmeisser, 2021-07-10:
Yes exactly, I can create the route like in the screenshot but then when I want to read anything I get the 1827

chrisbeardy, 2021-07-10:
Ok. This behaviour is unexpected. I suspect this may be a Beckhoff/TwinCat and or network type issue and not linked to pyads but we can't rule out that something is wrong with pyads. 

What version of TwinCat are you running? If you are on 4024 are you trying to set the route up with secure ads, try without. 

It might be worth contacting Beckhoff support to ask why access may be denied beyond just the wrong credentials. They probably won't be much help with the Ubuntu side but may be able to know what would cause this error between a two windows machines.  

Is this a base/standard Ubuntu and Windows setup, could there be some user permissions, network setup that could be causing issues. 

A common issue could be the firewall, are their exceptions for the TwinCat ports (these are usually added when installing twincat), I'd check for the TwinCat UDP and TwonCat TCP rules in the advanced firewall rules in Windows And then check for any firewall in Ubuntu.  Although as the first connection is made, I'm not sure this would be the issue. 

It could be worth checking that you can connect locally by running pyads on the windows PC where no route needs setting up and then trying to connect from a another Windows machine running python (if available). Are you on a Beckhoff IPC or is this a normal PC?  If it's your own, you could try using a virtual machine with a fresh install of Ubuntu and a bridged/host only network to see if you can connect from there too.

simonschmeisser, 2021-07-19:
I turns out I'm not supposed to set the `unidirectional` checkbox and then it works. Is there a good place to add this in the documentation? I opened a PR

Thanks for your help!

chrisbeardy, 2021-08-12:
thanks for the PR, I am sure the stlehmann will get around to it when he has some time. The route setup has always beens difficult to get right and explain. I am plannig on doing a video of it at some point when i get the time too.

stlehmann, 2021-10-16:
@simonschmeisser thanks for digging into it. Setting up the routes the right way is actually TwinCAT-related and is not directly covered by pyads. Especially if the routes are created with TwinCAT itself. Also the settings depend on the system you are on. E.g. If you are on Windows it might make perfectly sense to add a route remotely. But we can include a hint not to use the unidirectional setting.

commit 3c505092 (2021-11-09): Add hint on unidirectional routes in docs (#258)

* More explicitely explain route settings

fixes #255

* Change note