stlehmann/pyads #256: Segmentation fault when connecting with two clients

piertoni, 2021-07-09:
Hi,
I have a scripts that connects to the PLC, registers a callback on a symbol and polls for a value every 10 seconds (a sort of watchdog).
It works fine.
When I connect from another client in the script I get correctly this message:
`2021-07-09T15:57:18+0200 Info: connection closed by remote`

At this point I have the following issues:
1. I cannot know inside my script if the connection is still alive as `is_open` does only return a stored value
2. When I try to `plc.close()` before reopening (I need to do this to be able to communicate again) the software gives a Segmentation Fault. Note that happens if and only if I register a callback on a variable, if I don't register the software works as expected.

Any hint? Thanks

piertoni, 2021-07-10:
The problem I reported is on a Linux platform connecting to a Windows PLC

chrisbeardy, 2021-07-14:
hi, thanks for the issue, have you got a simplified code snippet to show what you are doing? this might help in debugging this issue.

piertoni, 2021-07-14:
Hi, this was reproduced also by another person, we also found how to circumvent it.

How to reproduce:
1. on a Linux system connect to a PLC and add a callback for a variable
2. open another shell on the same system and open another connection to the same PLC
3. When the second connection opens the first drops. At this point if the first connection does `plc.close()` you have the segfault.

Circumvent: you need to `del_device_notification` before close, it raises an exception because the connection is no more in place, but it will not segfault.

piertoni, 2021-07-14:
BTW: on Windows guest you can have multiple ADS connection at a time versus the same PLC, I understand that this is not possible on linux? (the first connection is dropped when you open another one)

chrisbeardy, 2021-08-12:
yes, thanks for finding a workaround, but yes I assume this is a bug. That is my understanding too that only one linux client can connect, although I do not know why. Sorry.