stlehmann/pyads #257: read_list_by_name generates "ADSError: parameter size not correct (1797)." but write_list works

ako-kft, 2021-07-13:
Stefan,  first of all: fantastic work! This little module got me right out of the blocks. 

I stumbled upon an issue I am unable to resolve:  

`plc = pyads.Connection(remote_ads_id, pyads.PORT_TC3PLC1, remote_ip)
plc.open()

read_list = ['GVL_Relais.k0','GVL_Relais.k1']
plc.read_list_by_name(data_names=read_list)
# leads to "ADSError: parameter size not correct (1797)."

write_dict = {'GVL_Relais.k0' : True, 'GVL_Relais.k1': True}
plc.write_list_by_name(write_dict)
# this works, I can turn relais on/off like this
`
read_by_name, write_by_name both work just fine. Any ideas what read_list_by_name is complaining about?

My host machine is Ubuntu 20.04, my PLC is a TwinCAT3 machine.

Thank you for your support! Best regards, Adrian

chrisbeardy, 2021-07-14:
I think this is the same issue as #232 . Therefore can i polielty ask you to close this issue and we can track it through #232.

ako-kft, 2021-07-14:
Agreed. -Adrian

closed 2021-07-14 by ako-kft