stlehmann/pyads #259: Read plc value depending on other value change

AndreSchm, 2021-08-17:
I would like to get the current value of a REAL variable in dependence of an other BOOL variable using a OnChange notification:
```python
@plc.notification(pyads.PLCTYPE_BOOL)
def callback(handle, name, timestamp, value):
    if value==True:
        print(f"handle: {handle}")
        print(f"name: {name}")
        print(f"timestamp: {timestamp}")
        print(f"value: {value}")
        print(plc.read_by_name(f"GVL_MyValues.MyRealValue", plc_datatype=pyads.PLCTYPE_REAL))```
```
Then I add this callback to my changing bool variable:
```python
plc.add_device_notification(f'MAIN.MyBoolValue', 
                                        pyads.NotificationAttrib(sizeof(pyads.PLCTYPE_BOOL)),
                                        callback)
```

Everything except the MyRealValue is printed when MyBoolValue changes to True. Do I have to configure something else or does pyads not support `read_by_name()` (I also tried `symbol.read()`) within a notification callback?

I am using pyads 3.3.4 on a Windows 10 machine.

chrisbeardy, 2021-08-19:
Hi there. The issue here is unfortunately you can not do another ads call (e.g. read_by_name) with the same plc object while within the callback function as the callback function happens in another thread. This normally results in an ads timeout issue. 

You have a couple of options. 

1) I have not actually tried this, but it may work so let me know if you do try.  you may be able to define a new connection within the callback function, e.g.

```python 
with pyads.Connection(plc.ams_netid, plc.ams_port) as plc2:
    print(plc2.read_by_name(...))
```

2) set a flag / add to an event queue etc and process the event in a main loop. This is normally the approach I take when I develop applications using pyads and PyQT. 

Hope this helps.

AndreSchm, 2021-08-23:
Thank you very much @chrisbeardy. The first option works fine.
Can you also please provide a small sample for the second option? Thanks a lot in advance.

chrisbeardy, 2021-08-23:
The example you show is essentially option 2. You are setting a flag, in this case one created for you by the symbol class, and then checking that flag in a loop. 

You could get more complicated using the queue feature etc. Search "communicating across threads in python" but it all depends what you are trying to do and how complex it is. If my option 1 works or your code works then great ha. 

Since this is a know limitation and you have a solution,if you could kindly close the issue on GitHub that would be great. Thanks.

closed 2021-08-24 by AndreSchm