stlehmann/pyads #281: Documentation on how to connect from Linux is not sufficient.

CreaM129129, 2021-11-01:
Hello,
First, let me know I appreciate your work on this module. It is very handy.

Unfortunately, I really struggle with connecting from Ubuntu. (On windows everything works fine)
It seems I am not the only one. There are many git-issues covering this topic. I have unsuccessfully gone through several of them, only to find more, without acutely solving the problem.

Please consider making a clear to follow cookbook on how to connect from Ubuntu.

chrisbeardy, 2021-11-01:
https://pyads.readthedocs.io/en/latest/documentation/routing.html

This is the routing section which contains an `adding routes to a target` section. 

This has been rewriten a few times and clearly is still not quite right. 

To aid us, could you please give some specific feedback about what you struggled with and what would have aided you.

I have plans to do a video showing how to connect from Windows and Linux. Would this be a good idea?

CreaM129129, 2021-11-01:
> https://pyads.readthedocs.io/en/latest/documentation/routing.html
> This is the routing section which contains an `adding routes to a target` section.

Yes, this is the page of documentation I struggle with.

> This has been rewriten a few times and clearly is still not quite right.
> 
> To aid us, could you please give some specific feedback about what you struggled with and what would have aided you.
> 
- I am confused by owerall order of chapters _(Creating routes on Linux and below)._ 
There are two notes and one warning. This makes me jump up and down in the text, quickly losing myself, even though text is very short. I think a less general example, but with clear order of instructions would help me more.

- For every constant eg. "SENDER_AMS", "PLC_IP"... I would love a note where can one get these values, why are they set and named like this and if there are some defaults values I shoud use (like "Administrator" and "1" for PLC_USERNAME and PLC_PASSWORD)

- Adding a section with common errors and their solution woud also be nice. Some common errors I found mutiple times:
-`Unexpected response from PLC: b"\x03f\x14q\x00\x00\x00\x00\x06\x00\x00\x80\xc0\xa8\x00\x1f\x01\x01\x10'\x01\x00\x00\x00\x01\x00\x04\x00\x04\x07\x00\x00"`
-`ADSError: ADSError: target port not found   ADS Server not started (6).`
-`Info: connection closed by remote`

> I have plans to do a video showing how to connect from Windows and Linux. Would this be a good idea?

Yes, video is a brilliant idea.

CreaM129129, 2021-11-01:
I finally managed to get the connection working. I decided to make this little "guide" on how to setup the routes.

Lets assume two computers on the same lan (I have disabled firewalls on both of them):

**Target PC**
- Runs windows
- Runs TwinCat3
- Locally simulates PLC (I did not test my aproach on a actuall harware)
- Has some ip, in my case `TARGET_PC_ID = "172.18.10.112"`

**Client PC**
- Runs Ubuntu
- Runs my Python script with pyads
- Has some ip, in my case 172.18.10.178

Goal is to read and change variables of PLC (running on Target PC) from Python script (Client PC)

**1. SET ROUTE ON TARGET PC, USING TWINCAT:**

- Navigate to TwinCAT -> Router -> Edit Routes:
[image]
- Add route, tick _Advanced Options_ and fill in details:
[image]
_Host Name / IP_ is ip of Client PC.
_Route Name_ does not matter.
_AmsNetid_ is ip of client PC and 1.1.
_Adress info_ is ip of client PC (probbaly does not matter?)
Dont forget to set Remote Route to None, untick Unidirectional, Target route to Static.
Than Add Route, I left Secure ADS (on pop-up window) unticked.

**2. Obtain TARGET_AMS_ID from Target PC:**
- Navigate to TwinCAT -> Router -> Change AMS NetId:
[image]
Copy displayed addres. We will use it in our python script.

**3. Finnaly our python script on Client PC:**
```
import pyads

pyads.open_port()
# CLIENT PC IP with .1.1 at the end (we added this when creating route via TwinCat):
CLIENT_AMS_NET_ID = "172.18.10.178.1.1"
pyads.set_local_address(CLIENT_AMS_NET_ID)
pyads.close_port()
# pass
# connect to plc and open connection
# route is added automatically to client on Linux, on Windows use the TwinCAT router
# we obtained TARGET_AMS_ID in step 2:
TARGET_AMS_ID = "192.168.0.31.1.1"
# regular ip address of target pc:
TARGET_PC_ID = "172.18.10.112"
plc = pyads.Connection(TARGET_AMS_ID, 851, TARGET_PC_ID)
plc.open()
# in MAIN in my PLC i have bool variable bRunOnlyOnce
b = plc.read_by_name("MAIN.bRunOnlyOnce")
print(b)
plc.write_by_name("MAIN.bRunOnlyOnce", False)
exit()
```

Don't forget to run your PLC.
As I discovered both TwinCAT and pyads very recently feel free to edit/correct this.
This [issue](https://github.com/stlehmann/pyads/issues/34) helped me to make everything working and create this post.

chrisbeardy, 2021-11-01:
We can look at adding a more step by step guide to create a route and possibly add a video too. Thanks for this.

kryskool, 2021-11-01:
Hi @CreaM129129 

In Windows, Twincat -> About Twincat,  give you the AmsId 

Regards,

mtroback, 2021-11-03:
I've tried to fix the documentation in this pull request: https://github.com/stlehmann/pyads/pull/278

kryskool, 2022-11-02:
@stlehmann can you activate Discussion feature on this repository,
this ticket can be move on discussion( eg: Tutorial category)

And Discussion can be a better for people ask for question

Regards,