stlehmann/pyads #287: Support read/write of arrays of structures

stlehmann, 2021-11-18:
Until now we can only read/write arrays of simple types or single structure variables. Reading/Writing an array of a defined structure would be a great enhancement

stlehmann, 2021-11-18:
This is already implemented.

chrisbeardy, 2021-11-18:
I was just about to cpmmemt on this saying this was already possible, maybe we should add to the docs how to do it?

stlehmann, 2021-11-18:
> [{key: value for key, value in item} for item in sequence["items"]]

Yes, that is how I found it. I was just looking at [these docs](https://pyads.readthedocs.io/en/latest/documentation/connection.html#structures-with-multiple-datatypes) before and there is no hint to it. When looking at the implementation I saw that it is already there. I'll reopen the issue and rename it so it adresses the docs.

mareksuscak, 2025-05-16:
@stlehmann can you share an example please? Does pyads support notifications for arrays of structures? Documentation indicates that it does not but is there any workaround?

chrisbeardy, 2025-05-16:
you can read and write arrays of structures but I do not think arrays work with notifications, even arrays of primitive types (e.g. just an array of INTS). You could however test this. I do not know of any workaround at the moment.