stlehmann/pyads #308: Adding the possibility to read a struct of struct

kbonnot, 2022-02-03:
At the moment, reading a struct value is possible thanks to 'read_structure_by_name', but it doesn't work for struct composed of structs:
[image]

I've tried adding this through a recursive function 

[image]

(my modification is l267), the num_of_bytes is correct, but the function for reading it can't cope.
[image]

It would be a greatly useful feature !

Thanks in advance

chrisbeardy, 2022-02-04:
There is a workaround for this if you define your structure in python with the other structure embedded: eg

```
TYPE test1 :
STRUCT
    bTest1 : BOOL;
    bTest2 : BOOL;
    bTestStruct : test2;
END_STRUCT
END_TYPE

TYPE test2 :
STRUCT
    bTest3 : BOOL;
    bTest4 : BOOL;
END_STRUCT
END_TYPE
```

```python
TEST_STRUCTURE_1: pyads.StructureDef = (
    ("bTest1", pyads.PLCTYPE_BOOL, 1),
    ("bTest2", pyads.PLCTYPE_BOOL, 1),
    ("bTestStruct.bTest3", pyads.PLCTYPE_BOOL, 1),
    ("bTestStruct.bTest4", pyads.PLCTYPE_BOOL, 1),
```

I have made a function which helps you combine two structures.

```python
def _amend_def_var(var, original_def: pyads.StructureDef) -> pyads.StructureDef:
    return tuple(((var + "." + i[0]),) + tuple(i[1:]) for i in original_def)
```

Unfortuantly there is no PLC_TYPE for a structure or way to recognise something is a structure using ADS, which means the read_structure_by_name is a "hack" which just converts a stream of bytes, we may be able to manipulate the function so you pass a list of structure definitions into the function then it could look up if no PLC type is there a matching definition passed to it that it can use for recursion.

tboegi, 2023-04-30:
...PLC_TYPE for a structure or way to recognise something is a structure using ADS

There is a definition: search for something call BIGTYPE (or similar)
See here, it is defined as 65:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcadsocx/12466639883.html&id=

However, I don't have a good pointer to "decode" the "bigtype" into the members of a struct. 
But it should be possible, since TwinCAT.exe itself is able to do so. More digging needed

FilipeJSRosa, 2023-07-19:
Any update on this item?
Still not possible to read a struct of struct without declaring them on python?

chrisbeardy, 2023-07-19:
The best update I can give is that this feature is not actively being worked on. This is a project run by volunteers so unfortunately development often happens in stints when the time allows. If you are familiar with python and twincat please feel free to submit a PR. Thanks

PR #384 (2024-08-30): Added feature for processing nested structures
Enables processing for nested structures, as mentioned in and fixes #308 and on [Stackoverflow](https://stackoverflow.com/questions/74134828/how-to-access-a-nested-structure-using-pyads). Takes into account nested structures in the following form:
```python
    substructure_def = (
         ("iVar", pyads.PLCTYPE_INT, 1),
         ("sVar", pyads.PLCTYPE_STRING, 1)
     )

    structure_def = (
         ("iVar1", pyads.PLCTYPE_INT, 1),
         ("structVar", substructure_def, 2)
     )
```
or
```python
    structure_def = (
        ("iVar1", pyads. [... cut]
closed 2024-08-30 by chrisbeardy