stlehmann/pyads #310: How can i check the current connection status to the PLC ?

FranciscoSantiago15, 2022-02-15:
Hi,

I m programming a Beckhoff CX7000 running with TwinCAT v3.1, and I m using the pyads library to stablish a connection. 

Is it possible to know the state of the connection between the plc and the script? that is, I would like to know if it is possible to monitor the connection to know when it is broken or signal failures may have occurred.

FranciscoSantiago15, 2022-02-16:
> Hi, maybe you can use `plc.read_state()` to monitor the connection state

Yes i m using that but i dont understand what are the differences between '(5, 0)' and '(6, 0)'.
And another case is when i disconnect the plc and try do the `plc.read_state()` i m always getting the '(5, 0)'

chrisbeardy, 2022-02-16:
read state corresponds to these https://infosys.beckhoff.com/english.php?content=../content/1033/tcadsdll2/html/tcadsdll_enumadsstate.htm&id=7003723284846371174

you can add a notifcation property for this using 

```python
self._plc.add_device_notification(
                (int("0xF100", 16), int("0x0000", 16)),  # ADSIGRP_DEVICE_DATA, ADSIOFFS_DEVDATA_ADSSTATE
                pyads.NotificationAttrib(ctypes.sizeof(pyads.PLCTYPE_INT)),
                self._on_plc_status_change,
            )

not usre what happens to this though if you physically lose connection to the PLC, i.e. the cable is pulled out

```

chrisbeardy, 2022-02-16:
this are also deines in the constants section of pyads, e.g. 

```python
pyads.ADSSTATE_RUN
```

closed 2022-02-18 by stlehmann