stlehmann/pyads #315: PyAds on PLC itself(Windows), cannot connect to PLC...

james089, 2022-04-13:
How can I connect to local PLC using PyAds when my Pyads is running on PLC itself???

If we write our code in C#, we don't have to add any route since it is the same PC, we create adsClient and connection using Port 851 only. 

But for PyAds running on PLC itself, I tried to do the same thing: 
1. I can't pass Port only(like, plc = pyads.Connection(851)) , it requires a AmsNetID. 
2. When I pass AmsNetID of the PLC, it shows "Failed to open port on AMS router".

james089, 2022-04-13:
Update: python version has to be < 3.10, tried 3.9.x, it works right away, but in 3.10, it doesn't work.

chrisbeardy, 2022-04-14:
I always do this:

```python
pyads.open_port()
ams_net_id = pyads.get_local_address().netid
pyads.close_port()
port = 851
plc = pyads.Connection(ams_net_id, port)
```

not sure why you couldn't connect on python 3.9, it should be compatible...

Can you please close this issue if your problem is solved, many thanks.

Summerxiaxia623, 2022-04-15:
I changed my python version from 3.10 to 3.9 and it works.

chrisbeardy, 2022-05-31:
@james089 or @stlehmann can we change the title of this issue to "Connecting to localhost not working on python 3.10" to easily differentiate this

RobertoRoos, 2022-10-13:
I have a fairly fresh install of Python 3.10 (running Windows 10) and I can connect to the local PLC without any problems.

closed 2023-10-20 by chrisbeardy