stlehmann/pyads #318: Error while "import pyads"

BlueShit-1120, 2022-05-01:
Hello,

When I using "pip install pyads" on my Client PC (Runs windows).  When I test my installation, an error happens:
`Traceback (most recent call last):
  File "C:/Users/xxwan/Desktop/py_keypoint/test_plc.py", line 1, in <module>
    import pyads
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\site-packages\pyads\__init__.py", line 10, in <module>
    from .ads import (
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\site-packages\pyads\ads.py", line 49, in <module>
    from .pyads_ex import (
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\site-packages\pyads\pyads_ex.py", line 69, in <module>
    _adsDLL = ctypes.WinDLL("TcAdsDll.dll")  # type: ignore
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] Could not be found the specified module .`

Do I need to do other things else after installation? Could please give me some advice, Thanks.

chrisbeardy, 2022-05-01:
Do you have twincat installed on the client pc? You need at least twincat ads installed on the client pc if its windows based

BlueShit-1120, 2022-05-02:
> Do you have twincat installed on the client pc? You need at least twincat ads installed on the client pc if its windows based

Thanks for your advice. I just installed Twincat 3 on the target PC. "import pyads" works normally after I install Twincat 3 on my client PC.

I met another error when I test the connection:
**Target PC**
1) runs windows,
2)runs Twincat3 and plc,
3)TARGET_PC_ID = "144.214.174.72" and TARGET_AMS_ID = "169.254.42.168.1.1"
**Client PC**
1) runs windows,
2)runs Twincat3,
3)Client_PC_ID = "144.214.174.69" and CLIENT_AMS_NET_ID = "144.214.174.69.1.1"

I have set the route on the target PC using Twincat. Here is my python script on Client PC:
`import pyads
TARGET_AMS_ID = "169.254.42.168.1.1"
TARGET_PC_ID = "144.214.174.72"
plc = pyads.Connection(TARGET_AMS_ID, 851, TARGET_PC_ID)
plc.open()
print("state",plc.read_state())`
But an error happened:
`Traceback (most recent call last):
  File "C:/Users/xxwan/Desktop/py_keypoint/test_plc.py", line 14, in <module>
    print("state",plc.read_state())
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\site-packages\pyads\connection.py", line 237, in read_state
    return adsSyncReadStateReqEx(self._port, self._adr)
  File "C:\Users\xxwan\AppData\Local\conda\conda\envs\tensorflow-gpu\lib\site-packages\pyads\pyads_ex.py", line 510, in adsSyncReadStateReqEx
    raise ADSError(error_code)
pyads.pyads_ex.ADSError: ADSError: target machine not found    Missing ADS routes (7). `

Does this mean the connection failed? Whether I need to add route on my client PC? I have checked other issues, no one else seems to have this problem. I need your advice. Thanks.

BlueShit-1120, 2022-05-02:
BTW, how to get the right port number? Thanks.

chrisbeardy, 2022-05-02:
I would add the route on the client pc, also please check your firewall for twincat. https://infosys.beckhoff.com/english.php?content=../content/1033/tcremoteaccess/html/tcremoteaccess_firewall.html&id=

BlueShit-1120, 2022-05-03:
> I would add the route on the client pc, also please check your firewall for twincat. https://infosys.beckhoff.com/english.php?content=../content/1033/tcremoteaccess/html/tcremoteaccess_firewall.html&id=

Thanks for your answer. It works now. I need to add route on both PC (Target and Client) and work in run mode.

closed 2022-05-03 by BlueShit-1120