stlehmann/pyads #319: Connection Closed By Remote

joshgraff-planted, 2022-05-16:
New user, it took me a few days to figure out connections.
I have a windows PC running TC3, connected over ethernet to a windows PC running windows Subsystem for Linux, with python version Python 3.8.10.
The TC3 ADS router is configured:
[image]

I get intermittent failures on my pyads connection to the PLC. 
2022-05-16T14:15:49-0700 Info: Connected to 192.168.10.145
2022-05-16T14:15:50-0700 Info: connection closed by remote

My entire code is attached, but the test script essentially calls the ADS system at irregular intervals 100 times. I found that on any given run, ~%20 fail in this way. The pattern is similar every time, the code will fail five or six times, and then succeed for a seemingly random number of attempts. I'm not quite sure what's going on, or where to look; I'm pretty sure my code is very similar to the examples. 

[test_pyads.txt](https://github.com/stlehmann/pyads/files/8703508/test_pyads.txt)
 
This is actually my first github issue, too (I've read many of them), so if I should change my strategy in the future, that would also be helpful.

tboegi, 2022-05-17:
This is just a guess/question:
How many ADS connections do you have from the client (the WSL box)
to the PLC ?
There can only by one: 
If you run TwinCAT.exe at the same time as you run
the pyads code: The ADS router on the PLC gets confused.
Solution: Close TwinCAT.exe while using pyads on the same machine.

joshgraff-planted, 2022-05-17:
Your comment makes me think of the issues I've been having executing my pyads code from the same PC (on localhost). I'll open another issue for that, though, as it doesn't seem to apply to this issue.

joshgraff-planted, 2022-05-17:
Interesting suggestion. My architecture is as below, one PC with WSL and Pyads connected to another with Windows and TwinCAT. 
PC (Windows, Pyads on WSL) - > PC (Windows, TwinCAT)
There isn't a TwinCAT.exe running on the twincat PC, not that I can see anyways; I have TCshell and the PLC execution service running though; I tried closing the shell/IDE and it didn't have an impact on the issue I show here.

One thing that did solve this issue was to not open a connection on every iteration, but instead to keep the connection open between iterations of the code. I'm running a test now, and it shows 0 failures for a few hours now (even with Timeout set lower than the delay between executions, so I'm not sure what exactly it measures to keep the connection open.) 

I've attached my code here in case someone comes across this issue in the future. 
[test_pyads.txt](https://github.com/stlehmann/pyads/files/8705212/test_pyads.txt)

closed 2022-05-17 by joshgraff-planted