stlehmann/pyads #330: Feature request: support TwinCAT automation interface

rerpha, 2022-10-13:
As someone who configures twincat devices I would like there to be a method in python to automate the process. Currently there are c++ and .NET interfaces to do this from c++/C#, but no method i am aware of from python. 

If this is considered out of scope for pyads no worries, feel free to close

docs: https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_adsnetref/7313118347.html&id=

chrisbeardy, 2022-10-13:
I think this is outside of scope of this library as the automation interface uses a completely different mechanism and has a different purpose. However I know of no other libraries that makes the AI interface available to python so it would be cool to start looking at it, maybe you could take a look if you already use the AI? I think all the calls are just COM calls, currently dont know how to do that using python but sure there is a way, could just wrote a wrapper using pythonnet maybe....

rerpha, 2022-10-13:
fair enough - will close this issue. I can't say I do either, but will do some digging.

closed 2022-10-13 by rerpha