stlehmann/pyads #333: Pyads connection refused with Beckhoff Twincat 3

rudy93, 2022-11-08:
Hello,

I am trying to make a connection from a server to a Beckhoff PLC with TwinCAT 3. With Windows everything works fine but with the same server on Linux I can't get a connection. The server has a static IP and in the route manager in the PLC I can find the route and see the server. I have tried adding the route by the route manager in the PLC and with "add_route_to_plc" but both ways my connection is refused. I have already turned off all firewalls. Any of you guys any idea what goes wrong here? In the attachment I have added some picture to see my settings and code.
[image]
[image]
[image]
[image]
[image]
[image]

rudy93, 2022-11-10:
It helps me a bit, I was already scared that it is probably a problem in Linux.... I don't like Windows as a server, that crashes to much in my opinion. Do you maybe know the best place to ask my question?

rudy93, 2022-11-13:
Just checked: 255.255.255.0, so should be OK

rudy93, 2022-11-22:
I don't know exactly how, but I solved the problem. We did everything step by step again, and it worked. But I don't need to close the connection every time. I will close the issue! Thanks for all the input.

closed 2022-11-22 by rudy93

AbdulFaiz123, 2024-02-07:
I am Trying to Do the connection as you did , but I am unable to connect I get the same Problem .,
If. you solved it Can you please Explain here