stlehmann/pyads #336: Running PyAds from Jetson Nano

cunhafh, 2022-11-19:
Hi, I'm trying to run PyAds on a Jetson Nano however I keep getting "ADSError: target port not found ADS server not started"

These are the steps that I followed so far:

- Installed python 3.7
- Installed pyads from source
- Added a route on my Windows machine that is running Twincat3 (4024.20). Remote Route = None / Server, AMS_NET_ID = 172.17.0.1.1.1, IP = 172.17.0.1, Unidirectional is unselected
- My Windows PC is in the same subnet and I can ping the Jetson nano from there.
- I run this code:

`import pyads
remote_ip = MY_PLC_IP
remote_ads = MY_PLC_AMS_NET_ID
pyads.set_local_address(MY_JETSON_AMS_NET_ID)
with pyads.Connection(MY_PLC_AMS_NET_ID, pyads.PORT_TC3PLC1, MY_PLC_IP) as plc:
    plc.read_by_name('.TAG_NAME', pyads.PLCTYPE_INT)`
    
Any ideas of what I could be doing wrong? Thanks

cunhafh, 2022-11-20:
Never mind, network issues!
In case you ever wondered, it works on the Jetson Nano.
Cheers

chrisbeardy, 2022-11-20:
Thanks @cunhafh , please close this issue if it works

closed 2022-11-21 by cunhafh