stlehmann/pyads #339: Read by name

pedromartinsbosch, 2022-12-21:
A few months ago I noticed a problem with the read_by_name function.
My problem is explained in this link: https://stackoverflow.com/questions/73691829/adserror-857212673-cant-read-variable-values-through-name-pyads

chrisbeardy, 2022-12-21:
Is this a TwinCAT 2 or 3 system? The traceback provided seems to be having an issue acquiring the handle. Are you sure the variable Test.M0 is the full path, should it be MAIN.Test.MO or GVL.Test.M0 etc?

pedroaccwinkelmartins, 2022-12-22:
This is a TwinCAT 2 system. I tried using the full path with the MAIN prg (Test is prg called in the MAIN prg), and it still did not work (gave the same error message)

chrisbeardy, 2022-12-22:
OK, I think if Test is a PRG then the correct path is Test.M0. Have you tried read by name when not giving it a fixed memory address? There may be a bug here somewhere. Can you also try with an older version of pyads, There have been a couple of TwinCat 2 related bugs more recently, and I wonder if somewhere compabitlbity broke with TwinCat 2.

pedroaccwinkelmartins, 2023-01-04:
I've tried both your suggestions (downgraded pyads to version 3.1.3 and stopped assigning a fixed memory), but in both cases I wasn't able to get the right results

chrisbeardy, 2023-01-04:
There may be a bug/ issue here to do with the overall set-up and using the BC9120. I also do not have a BC9120 to test anything on. This is not something I have encountered before, especially the issue with the unknown error. I apologise that we can not help any further. You may have to look at using the .NET or another Beckhoff supported ADS interface if you wish to look at reading the variables over ADS. It would be interesting to see if these actually work using read by name too.

As on last ditch attempt, you could try using the symbol interface to read the variable, as this has a slightly different mechanism to read and acquire the address.

https://pyads.readthedocs.io/en/latest/documentation/symbols.html

The BC9120 may not be supported by the C / Python ADS API.

Please can you change the title of the issue to "Unknown Error using Read by name on a BC9120". 

Thanks

m4tek, 2024-08-01:
I'm using bunch of BC9100s with this `pyads` library and I'm encountering the same issue since the beginning. 
It looks to me (not an expert though) that BC9xxx controllers do not maintain the symbol mapping at all and require reading variables by index+offset.
Even when working with terminals in `PLC terminal (complex)` mode and having them available to the local PLC, symbols tables are no maintained or I haven't found a way of accessing them.

chrisbeardy, 2024-08-02:
When you read by memory and offset does it work? Does using the pyads symbol work? https://pyads.readthedocs.io/en/latest/documentation/symbols.html