stlehmann/pyads #340: How to create PLCVariable of type list

tall-dan, 2022-12-24:
Hello, I'm trying to write some tests which mock my plc. One of the variables I'm using is an array of strings, but I'm having trouble mocking it:

```python
# Mock PLC:
test_server = AdsTestServer(logging=False, handler = AdvancedHandler(), port = port )
test_server.start()

test_server.handler.add_variable(PLCVariable(f'aBarcodes', bytes(), pyads.constants.ADST_STRING, 'STRING', 100)`

# client
plc = pyads.Connection('127.0.0.1.1.1', 48898)
plc.open()
plc.write_by_name('aBarcodes', ['1','2','3'], pyads.PLCTYPE_STRING)
# AttributeError: 'list' object has no attribute 'encode'
```

Any help would be much appreciated!

chrisbeardy, 2022-12-24:
I haven't checked as on mobile but are their any examples in the tests folder of doing this in pyads's own tests?

tall-dan, 2022-12-24:
Hi @chrisbeardy thanks for the quick response. I looked through the tests, specifically [`test_connection_class.py`](https://github.com/stlehmann/pyads/blob/master/tests/test_connection_class.py). There's a few examples of structs in there, but no lists

ChurchillSD, 2023-01-16:
If you haven't already found a solution, the way I achieved it was to create a byte array with the correct size (in this example, each element was 21 bytes, 20 for the chars in the string and a null terminator):

```python
expected_string_array = ["hello", "world", "", ""]

# Set up the String array
string_char_size = 20
string_element_size = string_char_size + 1
string_bytes = len(expected_string_array)*(string_element_size*[0])
for i, element in enumerate(expected_string_array):
    current_offset = string_element_size * i
    string_bytes[current_offset: ((len(element)) + 1)] = element.encode("utf-8") + b"\x00"

# Add to test plc
self.handler.add_variable(PLCVariable(
    name = "str_test_array", 
    value = bytes(string_bytes), 
    ads_type = constants.ADST_STRING, 
    symbol_type = f"STRING({string_char_size})"))```

Hope this helps

closed 2023-01-17 by tall-dan