stlehmann/pyads #341: Cannot read data from linux22.04 to window10

hovandieu, 2023-01-11:
I have created a program that reads data of twincat3 program running on windows10 using the program running at linux22.04
[image]
The router is created successfully and I have created a connection to the program in Twincat3 on my windows 10 machine.
However, when reading data from the program in Twincat3, the error is generated as below:
[image]
I have turned off the firewall on my windows10 machine but still can't read or write data for the program in Twincat3

chrisbeardy, 2023-01-11:
It looks like you are creating a connection to the wrong AMS net ID. You should use the NET ID of the PLC in pyads.Connection. 

e.g.

`plc = pyads.Connection("169.254.136.152", 851)`

hovandieu, 2023-01-12:
[image]

169.254.69.79 is IP connect to PLC device
169.254.136.152 is IP from router. In NUC pc run operation sytem linux 22.04 can ping to this ip.

> It looks like you are creating a connection to the wrong AMS net ID. You should use the NET ID of the PLC in pyads.Connection.
> 
> e.g.
> 
> `plc = pyads.Connection("169.254.136.152", 851)`

-> 173.31.128.1.1.1 is AMS NET ID of PLC. i think it correct with syntax of pyads.connection()

hovandieu, 2023-01-12:
In one network, i have NUC PC run linux 22.04 and create pyADS program to connect with Laptop Window 10 (run Twincat3 has AMS NETID = 173.31.128.1..1.1).
In laptop have 2 IP:
169.254.69.79 is IP connect to PLC device 
169.254.136.152 is IP get from router.

And in Laptop (Twincat3), result when i click Broadcast Search is nothing.

closed 2023-01-12 by hovandieu