stlehmann/pyads #343: Is pyads able to handle RPC methods and properties?

rako233, 2023-01-22:
I'm working heavily with classes. My vision is to mirror a class on the PLC in a HMI without defining the class in python. As long as references (Interface, Reference To) aren't read, this should be possible. A property will be transformed into 2 RPC methods by the compiler. 

`__get<property name>:<TYPE>`
`__set<property name(TYPE)>`

A class definition should be available in  the DATAINFO cmd of ADS. The reason for not wanting a definition in Python is, the info is already in place.

chrisbeardy, 2023-01-25:
Unfortunately not directly using direct methods in pyads, this is planned, see #190. 

However there will be a way of doing so by calling the CreateVariableHandle and ReadWrite functions yourself. This is just what the planned changes in #190 would do anyway.  You can follow the comment chain in #190 if you wish to implement this is your own program. If you do find the time and do this in your own application, please feel free to implement this feature into pyads, it will be appreciated!

Could you also please close this issue as it is effectively a duplicate of #190, any questions or comments, please feel free to comment on the issue #190.

Thanks

chrisbeardy, 2023-04-03:
@rako233 can you please close this issue to help manage the issue tracker, thanks.

chrisbeardy, 2023-05-09:
@rako233 @stlehmann this can be closed as duplicate of #190. Thanks.

closed 2023-08-30 by rako233