stlehmann/pyads #345: Connecting to PLC over ethernet

bunjopolo, 2023-03-16:
I am tying to read and write variables to an ethernet connected plc but I am getting the following error. 

pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861). 
2023-03-16T13:01:42-0700 Info: connection closed by remote

The PLC is connected over ethernet and I am able to ping it and pyads will extablish a connection but the error shows up when reading or writing any variables. Im using a Mac so im pretty sure I dont have to create a route like on windows from reading the documentation. 

When I get the local address it shows the address of my ethernet port that is connected to the plc and not the address of the PLC itself but i dont know why this would be. 

This is the output:

2023-03-16T13:01:37-0700 Info: Connected to 192.168.100.27
<AmsAddress 192.168.100.30.1.1:30000>

This is my code:
```python
import pyads

AMS_NET_ID = '192.168.100.27.1.1'
plc = pyads.Connection(AMS_NET_ID, pyads.PORT_TC3PLC1)
pyads.pyads_ex.adsAddRoute(AMS_NET_ID, ')
plc.open()
print(plc.get_local_address())
plc.read_by_name('Plywood.mybool')
plc.close()
```

chrisbeardy, 2023-03-17:
Hey, I'm not sure we officially support Mac but it may work due to its Unix routes. You still need to create a route, I would follow the docs for setting up a route in Linux and see how you get on

chrisbeardy, 2023-04-03:
@bunjopolo how did you get on? Can you please close this issue to help manage the issue tracker. Thanks.

bunjopolo, 2023-04-05:
Yes I was able to figure it out. One of the comments in this [Issue](https://github.com/stlehmann/pyads/issues/281) provides a guide for linux that worked for mac.

closed 2023-04-05 by bunjopolo