stlehmann/pyads #346: Connecting to simulated PLC

acolovini, 2023-03-30:
Hello,

I'm writing a client GUI application that will need to comunicate to a remote plc over ethernet. 
For the moment I would like to test it connecting a Laptop with the pyads software through local network to my PC that's running TwinCAT with the emulated PLC.

My goal for the moment would be to just connect to it and read/write some booleans.
In my network, the Laptop has IP 192.168.1.120 while my TwinCAT-running PC has IP 192.168.1.148

I've tried to connect directly using the AsmNetID 192.168.1.148.1.1 which is the on the PLC has, but i get this error:
ADSError: target machine not found    Missing ADS routes (7). 

If I try, however, to create a route on TwinCAT, I can't manage to make it work.

Since I'm a complete TwinCAT beginner, I'd like to know if this kind of connection is indeed possible and how I could set-up the route so that it works.

Thanks!

chrisbeardy, 2023-03-30:
Hello, thank you for your question, you are right that you need to create a route. Please take a look at the beckhoff documentation on how to do this or contact beckhoff support if there are issues.

I would check your firewall first, you may have to allow the right ports to the rules. Maybe turn it off first to check if this is the issue. 

Could I kindly ask you to close this issue as it is not related to pyads itself but rather TwinCat in general, thank you.

chrisbeardy, 2023-04-05:
@acolovini i hope you managed to solve your issue, please can you close this issue. I suggest posting on Stack Overflow with tags "plc". "twincat" or "twincat-ads" for other queries. Thanks.

chrisbeardy, 2023-05-09:
@acolovini @stlehmann this can be closed. Thanks.

closed 2023-10-20 by chrisbeardy