stlehmann/pyads #347: Failed to open port on AMS router.

JohnIsak, 2023-04-04:
I would appreciate some help to resolve this issue. 
I get this error message "Failed to open port on AMS router." both when calling pyads.open_port()  or when calling open() on a pyads.connection. This seems to indicate to me that the issue does not lie in the information provided to pyads.Connection but rather somewhere else, I don't know where.

chrisbeardy, 2023-04-05:
hello, thank you for your question, your first step is to check that the routing is set up correctly or if you are connecting to localhost you still provide an ams address and port number. This should be explained in previous issues and in the documentation.

May I ask that the issue tracker is reserved for issues on pyads and that questions are diverted to stackoverflow. Can I also ask that you search the previous issues before posting too, you will find that this topic has been discussed many times. 

Please can you close this issue as it is not an issue with pyads itself and if you are still struggling with getting a connection, please ask a question on stack overflow and tag it with twincat or twincat-ads and I will assist you.

Thanks.

JohnIsak, 2023-04-05:
I have no issues with getting a connection with the twincat software itself, both to a physical PLC or localhost. Based upon  your reply in #315, I don't think my issue is related to the AMS adress or port number since it seems you should be able to call `pyads.open_port()` regardless. Thanks for your help, I have no idea what the issue is exactly so I don't know the best place to ask.

chrisbeardy, 2023-04-05:
Are you trying to connect to localhost (running python and twincat on the same VM), or running just TwinCat on the VM?

If you are running python on a separate PC, is it windows or Linux? If windows you need to also have TwinCat installed on that PC, or at least the ADS only version. And you also need to create a route: https://pyads.readthedocs.io/en/latest/documentation/routing.html.

When do you get the error? Could you please post sample code.

JohnIsak, 2023-04-05:
I am running both Twincat and Python on the same VM.

I got the same issue both on python 3.9 and 3.11.

I get the error if I run the following sample code 

```
import pyads
pyads.open_port()
```
I ran 
Twincat\AdsApi\TcAdsTest
in order to find out more about the issue.

it seems that TcAdsTest is able to open a port: "AdsPortOpen: port = 32974"
AdsGetLocalAddr seems to work fine after opening a port.
"Test" gives me a warning "InitCombo: Registry key SOFTWARE\beckhoff\TwinCAT\Remote does not exist"  I am not sure if this is related or not.

I also installed the TwinCat ADS powershell module and as far as I can tell everything seems to be working fine there. I think that this issue may be something easy or obvious that I have overlooked, I am not that familiar with TwinCat or TCP/IP in general.

chrisbeardy, 2023-04-06:
Thats odd where the error occurs before even trying to create a connection.  this may be that the twincat install didnt complete correctly and python is struggling to find the router. . I'd check the ads DLL is in the correct place too. C: twincat/adsapi/tcadsdll/.

JohnIsak, 2023-04-08:
The ads DLL seems to be in the correct place, changing the path specified in the pyads source code also changes the error message I receive. I am not quite sure how to verify that everything is install correctly w.r.t Beckhoffs software as everything else seems to be working fine, I have for example already programmed a PLC.

JohnIsak, 2023-04-13:
Good to know that the issue is not related to the latest build or python version.
I will try later to redo the setup from scratch and see if that can help. I also downloaded and ran this example [https://infosys.beckhoff.com/english.php?content=../content/1033/tcinfosys3/126100793907171851.html&id=](url) which also contains the ADS open port argument, it seems to be working fine. I am also considering setting up a c++ environment and compile this sample code and see if it then also still works.

chrisbeardy, 2024-01-25:
closing as AMS issue has come up multiple times so consolidating to a single issue

Yaro7619, 2025-01-22:
Hi, @JohnIsak
Have you resolved this issue?
Could you share some thoughts?

> Good to know that the issue is not related to the latest build or python version. I will try later to redo the setup from scratch and see if that can help. I also downloaded and ran this example [...]