stlehmann/pyads #348: Auto-update rate of symbols

gbcarlos, 2023-04-12:
Hi everyone!

I'm reading out 10 symbols in a loop with 10Hz using `symbol.read()` but I'm getting the same error as in #35.

I just read that it would be enough to use `symbol.value()` instead of `read()` if I set `auto_update=True`. Now I am wondering at what rate the symbol is updated automatically if I set that property and if it is possible to change the rate.

chrisbeardy, 2023-04-12:
The default settings for auto update=True will update the value if it changes on any plc cycle, therefore there is no need to update the rate, it will be as fast as your plc cycle.

If you wish to receive notifications cyclically at a set rate you have to attach the notification manually and can set the notification period in ms and the notification attribute to server cycle. 

https://pyads.readthedocs.io/en/latest/documentation/symbols.html#device-notifications
https://pyads.readthedocs.io/en/latest/documentation/connection.html#device-notifications

Can I please ask in the future that all questions are directed to Stack overflow using the tags PLC, Twincat and Twincat-ads. There is a small community there that will answer your questions. This is to just keep the Issues log as pyads issues. Thank you. 

If you could kindly close this issue that would be great, thanks.

gbcarlos, 2023-04-12:
Thanks a lot for the fast reply, I was not aware of the community but will post there in the future.

closed 2023-04-12 by gbcarlos

gbcarlos, 2023-04-14:
Thanks for that idea, I will keep that in mind. For now reading them in roughly at PLC cycle is enough for my means but I will come back to it if the amount of read requests increase in the future.