stlehmann/pyads #349: SyntaxError in 'import pyads' for python 2.7 32 bit

pedroaccwinkelmartins, 2023-04-26:
I have a python version 2.7.16, 32 bit and tried to import pyads.
However I receive this message when I execute the python script:
```
Traceback (most recent call last):
  File "c:/Users/MAP9AV/Documents/import subprocess.py", line 4, in <module>
    import pyads
  File "C:\Python27\lib\site-packages\pyads\__init__.py", line 8, in <module>
    from .structs import AmsAddr, NotificationAttrib
  File "C:\Python27\lib\site-packages\pyads\structs.py", line 31
    def __init__(self, stAdsVersion: SAdsVersion) -> None:
                                   ^
SyntaxError: invalid syntax
```
What could it be from?

chrisbeardy, 2023-04-26:
Hello, sorry Python 2 support was removed in pyads version 3.2. If you wish to use pyads with python2 you will have to go back to prior to this version. If this is not a legacy project, I highly recommend updating to python3 as python 2 support from python itself is now no longer too. 

Please can you close this issue as we do not plan on supporting python2 anymore. Thanks.

pedroaccwinkelmartins, 2023-04-26:
Ok, thanks for the quick feedback

closed 2023-04-26 by pedroaccwinkelmartins