stlehmann/pyads #350: How to access a Nested Structs?

fieryWaters, 2023-04-28:
Hi, the documentation does not cover nested custom datatypes. Is this supported?

For example, if I have:
TYPE TPoint : 
	STRUCT
		x : REAL;
		y : REAL;
	END_STRUCT
END_TYPE

TYPE ListOfPoints:
	STRUCT
		rampPoints : ARRAY [1..30] OF TPoint;
		rampSize : INT := 0;
	END_STRUCT
END_TYPE

In PyADS, I try: 

TPoint_def = (
         ("x", pyads.PLCTYPE_REAL, 1),
         ("y", pyads.PLCTYPE_REAL, 1)
     )

ListOfPoints_def=(
    ("rampPoints", TPoint_def, 30),
    ("rampSize", pyads.PLCTYPE_INT,1)
    )

test1 = plc.get_symbol("Main.test1", structure_def=ListOfPoints_def, array_size=1)

Is there a Pythonic way to do this or do I just need to avoid custom nested types? Thanks!

chrisbeardy, 2023-04-28:
hello, unfortunately this is not directly supported, but there is a workaround. please see #308. Hopefully support for this will be directly added to pyads at some point.

Please can you close this issue as it is a duplicate of #308, thanks. Any comments / additions, please add to #308.

chrisbeardy, 2023-05-09:
@stlehmann this can be closed, duplicate #308, thanks

closed 2023-10-20 by chrisbeardy