stlehmann/pyads #351: How to access 2D Arrays?

fieryWaters, 2023-04-28:
Hello, does pyads support accessing a 2D array directly? I didn't see anything in the documentation. Thanks!

chrisbeardy, 2023-04-28:
I know you can read an array of an array

```python
plc.read_by_name('GVL.sample_array[0][0]', pyads.PLCTYPE_INT)
```

you could try the following for multidimensional arrays:
```python
plc.read_by_name('GVL.sample_array[0,0]', pyads.PLCTYPE_INT)
```

Let me know how you get on

fieryWaters, 2023-05-06:
Tested works:

Declaration ST: 
	test3dArray : ARRAY [1..5, 1..5, 1..5] OF REAL;

Simple Access:
value = plc.read_by_name('Main.test3dArray[1,1,1]', pyads.PLCTYPE_REAL)

General Implementation:
#works for arrays of any number dimensions
def readArray(moduleName, variableName, *arrayDimensions ):
    asset = moduleName + '.' + variableName
    asset += '[' + ",".join(str(dimension) for dimension in arrayDimensions) + ']'
    return plc.read_by_name(asset, pyads.PLCTYPE_REAL)

#test readArray
value = readArray("Main", "test3dArray" , 1,1,1)

closed 2023-05-06 by fieryWaters

fieryWaters, 2023-08-09:
The above code works, but only for a single element, and is very slow when writing large arrays, since each write requires a unique query to the network. The faster way to transfer a large array to/from is to do the entire thing in one query. Below is how to accomplish this.

in MAIN POU on PLC:
```
VAR
	testArray : ARRAY [1..10,1..10, 1..10] OF REAL;
END_VAR
```

in PLC Python:
```
import numpy as np
plc = pyads.Connection('5.5.5.5.1.1', pyads.PORT_TC3PLC1)
plc.open()

array = np.linspace(0,10*10*10 - 1,10*10*10).reshape((10,10,10)) #generate 10x10x10 array in format of [1,2,3,...,999]
plc.write_by_name('Main.testArray', array.flatten(), pyads.PLCTYPE_REAL * 1000)

loopback = plc.read_by_name('Main.testArray', pyads.PLCTYPE_REAL * 1000)
loopback = np.reshape(loopback, (10,10,10))
print(loopback)
```
 If y'all want to update the documentation go ahead. Thanks.