stlehmann/pyads #359: Communication with Beckhoff virtual machine from another virtual machine

X-Carolinii-X, 2023-09-17:
Hello guys :)
I I try to connect and read data from plc which is running in TwinCat3 BSD machine. I want to connect from another virtual machine  (Linux Ubuntu) using python and pyads module. Here is all stuff and information:

TwinCat3 BSD machine:
[image]

Search adapters information from TwinCat3 on Windows:
[image]

Connection information from TwinCat3 on Windows:
[image]

Code in python on virtual machine with Linux Ubuntu:
```
import pyads

if __name__ == "__main__":
    remote_plc_ip = '192.168.2.68'
    remote_plc_net_id = '39.215.159.161.1.1'

    SENDER_AMS = '192.168.56.1.1.1'
    HOSTNAME = '192.168.56.1'
    PLC_USERNAME = 'Administrator'
    PLC_PASSWORD = '1'
    PLC_IP = '192.168.2.68'

    pyads.open_port()
    pyads.set_local_address(SENDER_AMS)
    pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, PLC_USERNAME, PLC_PASSWORD)

    plc = pyads.Connection(remote_plc_net_id, pyads.PORT_TC3PLC1, remote_plc_ip)
    plc.open()

    if plc.is_open:
        print('Plc is open!')

    try:
        plc.read_state()
        print('Connection succeeded')
    except Exception:
        print('Connection failed')

    plc.close()
```

And now when i try run python code i get:
```
Error: Connect TCP socket failed with: 111
Traceback (most recent call last):
  File "/home/carolinii/Projects/energy_institute_scada/scada-server-side/device_servers/ads/test_ads_connection.py", line 18, in <module>
    plc.open()
  File "/home/carolinii/.local/lib/python3.10/site-packages/pyads/connection.py", line 198, in open
    adsAddRoute(self._adr.netIdStruct(), self.ip_address)
  File "/home/carolinii/.local/lib/python3.10/site-packages/pyads/pyads_ex.py", line 167, in wrapper
    return fn(*args, **kwargs)
  File "/home/carolinii/.local/lib/python3.10/site-packages/pyads/pyads_ex.py", line 189, in adsAddRoute
    raise ADSError(error_code)
pyads.pyads_ex.ADSError: ADSError: target port not found   ADS Server not started (6). 
```

Anyone have idea what is wrong?

KrystianTolloczko, 2023-09-17:
1. Disable firewall in TwinCAT BSD:

`doas service pf stop `

Note: The firewall is automatically re-enabled after each restart. This behavior is ensured by the entry pf_enable="YES" in the rc.conf file.

2. Check and adjust ADS routes manually in TwinCAT BSD using StaticRoutes.xml 
[image]

3. Try this:
` plc = pyads.Connection(remote_plc_net_id, pyads.PORT_TC3PLC1)`

Hope this will help!

chrisbeardy, 2023-10-03:
@X-Carolinii-X if this fixed your issue, please can you close this issue to help us manage the issues log, thank you

closed 2023-10-20 by chrisbeardy