stlehmann/pyads #365: pip install pyads to aarch64 failed

toillone, 2023-11-16:
Hello

I encontered an issue after installed pyads using pip.
`pi@RevPi106327:~ $ pip install pyads
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Keyring is skipped due to an exception: Failed to create the collection: Prompt dismissed..
Collecting pyads
  Using cached https://www.piwheels.org/simple/pyads/pyads-3.3.9-py3-none-any.whl (330 kB)
Installing collected packages: pyads
Successfully installed pyads-3.3.9
pi@RevPi106327:~ $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyads
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/pyads/__init__.py", line 10, in <module>
    from .ads import (
  File "/home/pi/.local/lib/python3.9/site-packages/pyads/ads.py", line 49, in <module>
    from .pyads_ex import (
  File "/home/pi/.local/lib/python3.9/site-packages/pyads/pyads_ex.py", line 89, in <module>
    _adsDLL = ctypes.CDLL(adslib)
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/pi/.local/lib/python3.9/site-packages/pyads/adslib.so: wrong ELF class: ELFCLASS32
pi@RevPi106327:~ $ arch
aarch64`

After followed tutorial to install from sources it worked well

chrisbeardy, 2023-11-17:
looks like the wheel being built for the pi builds with 32bit adslib.so. Will need to get this fixed. In the meantime you will need to build from source. sorry.

chrisbeardy, 2024-07-30:
possibly fixed by #363 will close and reopen if reported again

closed 2024-07-30 by chrisbeardy