stlehmann/pyads #368: ImportError due to ELFCLASS32 Error with pyads on Raspberry Pi 64-bit

apku04, 2024-02-07:
I've installed pyads version 3.3.9 on my Raspberry Pi, running a 64-bit Linux kernel (6.1.0-rpi7-rpi-2712) and Python 3.11.2. However, when trying to import pyads, I encounter an OSError related to the ELF class of the adslib.so library:

   ` OSError: /usr/local/lib/python3.11/dist-packages/pyads/adslib.so: wrong ELF class: ELFCLASS32`

This error suggests that adslib.so is compiled for a 32-bit architecture, while my system is a 64-bit (aarch64). Given the architecture mismatch, pyads is unable to load the necessary shared library.

Has anyone encountered this issue and successfully resolved it by either finding a 64-bit version of adslib.so or compiling it for 64-bit architecture?

**Proposed Solution:**

I'm looking for guidance on how to either:

Obtain a 64-bit compatible version of adslib.so for pyads.
Compile adslib.so from source for a 64-bit (aarch64) architecture, if source code is available and this is a feasible solution.
Additionally, I welcome any advice on steps or considerations for compiling libraries on the Raspberry Pi 64-bit OS to ensure compatibility with pyads.

chrisbeardy, 2024-02-08:
Possibly linked to #363

chrisbeardy, 2024-02-08:
also linked to #22 #36 #147 #365, closing as duplicate of #365

closed 2024-02-08 by chrisbeardy

apku04, 2024-02-08:
ahh great...i'll follow the other issue then ..thx :)