stlehmann/pyads #380: docs not auto-building on readthedocs

chrisbeardy, 2024-04-01:
Hi @stlehmann ,

The readthedocs doesn't seem to have auto rebuilt upon releasing a new version or on any commit in the last 2 months. Can you please either:

- manually rebuild the docs on readthedocs
- check the readthedocs github integration / webhook
- or add me as an admin on the readthedocs pyads (if possible), and I can sort it out

Thanks

stlehmann, 2024-04-02:
Hey @chrisbeardy,

I triggered the docs build manually and got an error saying it can not find the pyads package which is weird because this has never been an issue before.

```bash
Read the Docs build information
Build id: 23936929
Project: pyads
Version: latest
Commit: 02e3815660ce7c0a5a5b48b545f6141e72d820df
Date: 2024-04-02T04:53:42.599582Z
State: finished
Success: False

[rtd-command-info] start-time: 2024-04-02T04:53:43.091159Z, end-time: 2024-04-02T04:53:43.428442Z, duration: 0, exit-code: 0
git clone --depth 1 https://github.com/stlehmann/pyads.git .
Cloning into '.'...

[rtd-command-info] start-time: 2024-04-02T04:53:43.446754Z, end-time: 2024-04-02T04:53:44.011191Z, duration: 0, exit-code: 0
git fetch origin --force --prune --prune-tags --depth 50 refs/heads/master:refs/remotes/origin/master
From https://github.com/stlehmann/pyads
 * [new tag]         3.3.5      -> 3.3.5
 * [new tag]         3.3.6      -> 3.3.6
 * [new tag]         3.3.7      -> 3.3.7
 * [new tag]         3.3.8      -> 3.3.8
 * [new tag]         3.3.9      -> 3.3.9

[rtd-command-info] start-time: 2024-04-02T04:53:44.099975Z, end-time: 2024-04-02T04:53:44.161892Z, duration: 0, exit-code: 0
git checkout --force origin/master
Note: switching to 'origin/master'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 02e3815 Update installation.md

[rtd-command-info] start-time: 2024-04-02T04:53:44.189431Z, end-time: 2024-04-02T04:53:44.259382Z, duration: 0, exit-code: 0
git clean -d -f -f

[rtd-command-info] start-time: 2024-04-02T04:53:44.297461Z, end-time: 2024-04-02T04:53:44.358464Z, duration: 0, exit-code: 0
cat .readthedocs.yaml
# Read the Docs configuration file for Sphinx projects
# See https://docs.readthedocs.io/en/stable/config-file/v2.html for details

# Required
version: 2

# Set the OS, Python version and other tools you might need
build:
  os: ubuntu-22.04
  tools:
    python: "3.10"

# Build documentation in the "docs/" directory with Sphinx
sphinx:
  configuration: doc/conf.py

# declare the Python requirements required to build your documentation
[... 707 lines cut ...]
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/envs/latest/lib/python3.10/site-packages/sphinx/application.py", line 216, in __init__
    self.config = Config.read(self.confdir, confoverrides or {}, self.tags)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/envs/latest/lib/python3.10/site-packages/sphinx/config.py", line 172, in read
    namespace = eval_config_file(filename, tags)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/envs/latest/lib/python3.10/site-packages/sphinx/config.py", line 341, in eval_config_file
    raise ConfigError(msg % traceback.format_exc()) from exc
sphinx.errors.ConfigError: There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/envs/latest/lib/python3.10/site-packages/sphinx/config.py", line 328, in eval_config_file
    exec(code, namespace)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/checkouts/latest/doc/conf.py", line 68, in <module>
    from pyads import __version__
ModuleNotFoundError: No module named 'pyads'

Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/envs/latest/lib/python3.10/site-packages/sphinx/config.py", line 328, in eval_config_file
    exec(code, namespace)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pyads/checkouts/latest/doc/conf.py", line 68, in <module>
    from pyads import __version__
ModuleNotFoundError: No module named 'pyads'
```

stlehmann, 2024-04-02:
@chrisbeardy I'd like to add you as an collaborator to the readthedocs project. Can you send me your email or RTD username?

Cheers, Stefan

chrisbeardy, 2024-04-02:
Its linked to my Github so the user name is the same as there. Hopefully that's enough...

chrisbeardy, 2024-04-02:
for my own record, tried downgrading the python version as that was first thing that appeared on google, didnt work, will try to build locally now and see if issue with sphink or readthedocs

stlehmann, 2024-04-02:
@chrisbeardy it could be necessary to install pyads first by doing `pip install -e .`

chrisbeardy, 2024-04-02:
Possibly, but it looks like an issue with either Sphinx or read the docs from my research as read the docs have changed there build process, in going to see if I can build locally correctly first to make sure that works then I can try and get readthe docs to replicate that

stlehmann, 2025-06-06:
Finally solved this one. Turned out the adslib submodule has not been cloned, thus opening the adslib.so failed.

PR #464 (2025-06-06): Fix readthedocs build process
This fixes #380
closed 2025-06-06 by stlehmann