stlehmann/pyads #391: I finally managed to get the connection working. I decided to make this little "guide" on how to setup the routes.

AndyEverything, 2024-06-19:
I finally managed to get the connection working. I decided to make this little "guide" on how to setup the routes.

Lets assume two computers on the same lan (I have disabled firewalls on both of them):

**Target PC**
- Runs windows
- Runs TwinCat3
- Locally simulates PLC (I did not test my aproach on a actuall harware)
- Has some ip, in my case `TARGET_PC_ID = "172.18.10.112"`

**Client PC**
- Runs Ubuntu
- Runs my Python script with pyads
- Has some ip, in my case 172.18.10.178

Goal is to read and change variables of PLC (running on Target PC) from Python script (Client PC)

**1. SET ROUTE ON TARGET PC, USING TWINCAT:**

- Navigate to TwinCAT -> Router -> Edit Routes:
[image]
- Add route, tick _Advanced Options_ and fill in details:
[image]
_Host Name / IP_ is ip of Client PC.
_Route Name_ does not matter.
_AmsNetid_ is ip of client PC and 1.1.
_Adress info_ is ip of client PC (probbaly does not matter?)
Dont forget to set Remote Route to None, untick Unidirectional, Target route to Static.
Than Add Route, I left Secure ADS (on pop-up window) unticked.

**2. Obtain TARGET_AMS_ID from Target PC:**
- Navigate to TwinCAT -> Router -> Change AMS NetId:
[image]
Copy displayed addres. We will use it in our python script.

**3. Finnaly our python script on Client PC:**
```
import pyads

pyads.open_port()
# CLIENT PC IP with .1.1 at the end (we added this when creating route via TwinCat):
CLIENT_AMS_NET_ID = "172.18.10.178.1.1"
pyads.set_local_address(CLIENT_AMS_NET_ID)
pyads.close_port()
# pass
# connect to plc and open connection
# route is added automatically to client on Linux, on Windows use the TwinCAT router
# we obtained TARGET_AMS_ID in step 2:
TARGET_AMS_ID = "192.168.0.31.1.1"
# regular ip address of target pc:
TARGET_PC_ID = "172.18.10.112"
plc = pyads.Connection(TARGET_AMS_ID, 851, TARGET_PC_ID)
plc.open()
# in MAIN in my PLC i have bool variable bRunOnlyOnce
b = plc.read_by_name("MAIN.bRunOnlyOnce")
print(b)
plc.write_by_name("MAIN.bRunOnlyOnce", False)
exit()
```

Don't forget to run your PLC.
As I discovered both TwinCAT and pyads very recently feel free to edit/correct this.
This [issue](https://github.com/stlehmann/pyads/issues/34) helped me to make everything working and create this post.

_Originally posted by @CreaM129129 in https://github.com/stlehmann/pyads/issues/281#issuecomment-956370290_

AndyEverything, 2024-06-19:
Many thanks for the detailed instructions. I have a Linux PC (Ubuntu) and would like to connect to a Beckhoff. The firewalls are off on both devices and the PLC is in run mode. I have created the route from the PLC to the PC manually according to the instructions. 

I run my Python script on the Linux PC: 

    import pyads
    TARGET_AMS_ID = "5.97.120.143.1.1"
    TARGET_PC_ID = "192.168.0.1"

    plc = pyads.Connection(TARGET_AMS_ID, pyads.PORT_TC3PLC1, TARGET_PC_ID)
    plc.open()
    device_name, version = plc.read_device_info()
    print(str(device_name) + ' ' + str(version))
    print(f"Connected?: {plc.is_open}")
    print(f"Local Address? : {plc.get_local_address()}")
    print(plc.read_state())

    plc.read_by_name("MAIN.test", pyads.PLCTYPE_INT)
    exit()

I receive the following feedback:

    ~$ python3 pyads_test.py
    2024-06-19T07:44:04+0000 Info: Connected to 192.168.0.1
    Plc30 App <pyads.structs.AdsVersion object at 0x7f67198859d0>
    Connected?: True
    Local Address? : <AmsAddress 192.168.0.253.1.1:30000>
    (5, 0)
    2024-06-19T07:44:04+0000 Info: connection closed by remote

For me, it looks like I can make a connection. Otherwise I wouldn't be able to read out the status. I could also create the route via add_route_to_plc. In this case, I have done it manually again. Nevertheless, the connection is closed by romete. Do you have a clue as to what the problem could be?

chrisbeardy, 2024-06-19:
Yes, it looks like you are creating a connection ok as you can ready the running state of the PLC (5 is PLC run). I am not sure what is now causing this to disconnect. What versions of python, twincat, ubuntu and pyads are you using? Are there any firewalls set up?

AndyEverything, 2024-06-19:
I have an Ubuntu 18.04.6 LTS with
- Python 3.7.17
- PyADS 3.4.0

The PLC has TwinCAT 3.1.4024.29.
I have deactivated the firewall on the Beckhoff and I have not set any special rules on the Linux.

I started a [Testserver](https://pyads.readthedocs.io/en/latest/documentation/testserver.html) on the Linux PC according to the documentation and was able to connect to it without any problems.

chrisbeardy, 2024-06-19:
I can't see anything wrong with that setup, you could try using an older version of pyads, we have recently changed how we are doing the build on Linux, so that may have untoward issues.

AndyEverything, 2024-06-20:
I have installed pyads on the Linux via **pip**. Since the installation and import works, I left it like this for a start. I should install it from source as recommended in the documentation.

heinz-raja, 2024-07-10:
In my case the problem was pyads could only add insecure routes to the target (AFAIK) and the target PC was configured to only accept secure routes also my target PC was not a windows machine but was running TCBSD so I couldn't add secure routes like mentioned in the above comment. I manually had to edit the StaticRoutes.xml file and add the tag attribute <RemoteConnections SecureOnly="False"> and then it worked.

closed 2024-08-13 by chrisbeardy