stlehmann/pyads #396: read_write()

albertzfg, 2024-07-19:
i am using python 3.12 and pyads 3.4

when call read_write() function, there is an error
'int' has no attribute if size

the code as below:

read_data_type = pyads.PLCTYPE_USINT * self.getReadSize()
        write_data_type = lambda v: (pyads.PLCTYPE_USINT * len(self.cmd))(*v)
        self.result = bytearray(self.conn.read_write(
            self.ADSIGRP_SUMUP_READ, self.n_vars,
            read_data_type,
            self.cmd, write_data_type))

i print out the two variables 
read_data_type
<class 'lib.gom_cx.c_ubyte_Array_260'>
write data type
<class 'lib.gom_cx.c_ubyte_Array_12'>

i debug into read_write()
looks like it comes from this line.

[image]

chrisbeardy, 2024-07-19:
thanks for this, can you please post a minimal code snippet which can replicate this issue? In your example it is not clear what self.cmd, self.getReadSize() etc is doing? What are you trying to achieve overall? it is normally not required for the user to call read_write directly?

albertzfg, 2024-07-20:
[image] 

this is how to prepare for self.cmd 
please help check  is that OK for read_write()parameter 
when i want to execute sumupread function

chrisbeardy, 2024-07-22:
OK, I'm not 100% sure what could be causing your error but I suspect at some point, python thinks that your self.cmd is now an int and not a bytearray before you call the read_write function?. Hence the error `'int' has no attribute of size`. Are you just trying to read multiple variables? Have you tried using the `read_list_by_name` method?

albertzfg, 2024-07-22:
so  in pyads what kind of format need to receive as value argument in read_write()function?

albertzfg, 2024-07-22:
> OK, I'm not 100% sure what could be causing your error but I suspect at some point, python thinks that your self.cmd is now an int and not a bytearray before you call the read_write function?. Hence [...]

actually the self.cmd is bytearray , but inside pyads_ex.py
 ```  
if index_group == ADSIGRP_SUMUP_READ:
        
        response_size = 4 * len(value)
        print(value)
        for _ in value:
            response_size +=_.size 
  ```
here is the `_.size` function reports the error "int has no attribute of size". 
so i am wondering what kind of format i need to provide to the read_write()function

chrisbeardy, 2024-07-30:
closing as this is a support, not an issue, planning on pening discussions in Github

closed 2024-07-30 by chrisbeardy