stlehmann/pyads #404: Issues Reading and Writing PLC BIT Type

keenanjohnson, 2024-07-31:
Hey all!

I have a PLC variable of type BIT.

However when I try to read and write to it with the pyads library I get some errors

For example the code below throws the error below:

Perhaps I'm doing something incorrectly? I've tried writing both Python bool values and numeric values to the same result.

### Code

```
en_symbol = plc.get_symbol('device.ch1')
print('Symbol Info: 'en_symbol)
print('Symbol Type: ', en_symbol.symbol_type)

# Try to write the Channel
en_symbol.write(True)
```

### Error

```
<pyads.symbol.AdsSymbol object at 0x7ffffea47470, name: device.ch1, type: BIT>
En channel 1 value:  BIT
Traceback (most recent call last):
  File "/workspaces/main.py", line 108, in <module>
    en_symbol.write(True)
  File "/usr/local/lib/python3.12/site-packages/pyads/symbol.py", line 221, in write
    self._plc.write(self.index_group, self.index_offset, new_value, self.plc_type)
  File "/usr/local/lib/python3.12/site-packages/pyads/connection.py", line 292, in write
    return adsSyncWriteReqEx(
           ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyads/pyads_ex.py", line 636, in adsSyncWriteReqEx
    data = plc_data_type(value)
           ^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable
2024-07-31T20:24:31+0000 Info: connection closed by remote
Segmentation fault
```

m4tek, 2024-08-01:
Can you share complete code snippet you are trying to run?

> print('Symbol Info: 'en_symbol)

This line contains a typo, there is a coma missing.

Generally there is no symbol type `BIT`, there is `PLCTYPE_BOOL`.

keenanjohnson, 2024-08-05:
So sorry yes that code snippet is complete minus the typo there. 

```
en_symbol = plc.get_symbol('device.ch1')
print('Symbol Info: 'en_symbol)
print('Symbol Type: ', en_symbol.symbol_type)

# Try to write the Channel
en_symbol.write(True)
```

There does indeed seem to be a bit type when I use the beckhoff tools to inspect the target. Perhaps the bit type is not exposed via ADS?

See the screenshot of the target browser of my PLC

[image]

chrisbeardy, 2024-08-13:
PLC type BIT is currently not supported directly, **however** if you specify the type as pyads.PLCTYPE_BYTE it reads them just fine, I don't know if you can actually read a bit directly over ADS (or if it then makes it a byte). I will have to experiment.

e.g. plc.read_by_name("device.ch1", pyads.PLCBYTE_BOOL) or symbol = plc.get_symbol('device.ch1'), symbol.plc_type = pyads.PLCTYPE_BYTE