stlehmann/pyads #421: Port: 30000 already in use

amirtronics, 2024-09-13:
I have a class with a plc member. I have a thread that sends a counter to MAIN.iCamClk. I have another node which writes to a struct. The codes works well without the clock thread but as soon as I try to run the thread I get:

```
2024-09-13T13:52:04-0700 Warning: Port: 30000 already in use as 0x7f282fffdf80
```

chrisbeardy, 2024-09-14:
Hi can you please provide a sample of your code that can reproduce the error, thanks.

amirtronics, 2024-09-16:
Here's my python code:

```
import threading
import pyads
import time

REMOTE_ADS = "X.X.X.X.1.1"
PLC_PORT = 851
REMOTE_IP = "X.X.X.X"

class PlcServer(threading.Thread):
    def __init__(self):
        super().__init__()
        self._stop_event = threading.Event()
        print("PLC Server initiated!")

        self.plc = pyads.Connection(REMOTE_ADS, int(PLC_PORT), REMOTE_IP)
        self.clkCounter = 0
        self.sequence = 0

        
        self.structure_def = (
                ("arrSeqData", pyads.PLCTYPE_INT, 2),
                ("bActivated", pyads.PLCTYPE_BOOL, 1)       
            )
        
        try:
            self.plc.open()
            print(f"PLC connected at AMS Address {REMOTE_ADS}")
            time.sleep(1)
            self.stLinuxData     = self.plc.get_symbol("MAIN.stLinuxData", structure_def=self.structure_def,  array_size=1)
            self.iLinuxClk       = self.plc.get_symbol("MAIN.iLinuxClk")
            self.isPlcConnected = True

        except Exception as inst:
            d = inst
            print(d)
            print(f"AMS Address {REMOTE_ADS} not found!")
            self.isPlcConnected = False

    def send_cmd(self):
        if self.isPlcConnected:
            try:
                self.stLinuxData.write([
                    {
                        "arrSeqData": [self.sequence, self.sequence],
                        "bActivated": True
                        }
                    ])
                self.sequence += 1
                        
            except:
                print(f"Connection lost with {REMOTE_ADS}!")
                print("Trying to reconnect ...")
                self._reconnect()
                time.sleep(0.5)

    def _reconnect(self):
        try:
            self.plc.close()
            self.plc.open()
            self.plc.read_state()
            self.isPlcConnected = True
            print(f"PLC reconnected at AMS Address {REMOTE_ADS}")
        except:
            self.isPlcConnected = False

    def stop(self):
        self._stop_event.set()

    def stopped(self):
        return self._stop_event.is_set()
    
    def run(self):
        while not self._stop_event.is_set():
            self.clkCounter += 1
            time.sleep(0.005)
            self.iLinuxClk.write(self.clkCounter)
            if self.clkCounter == 60000:
                self.clkCounter = 0

    def __del__(self):
        try:
            self.plc.close()
            if self.is_alive():
                self.join()
        except:
            pass

if __name__ == "__main__":

    plc_server = PlcServer()
    plc_server.start()

    try:
        while True:
            plc_server.send_cmd()
            time.sleep(0.5)

    
    except KeyboardInterrupt:
        print("Shutting down")

    finally:
        plc_server.stop()
        plc_server.join()

```

And here's the Plc stuct ST_LinuxData:

```
{attribute 'pack_mode':='1'}
TYPE ST_LinuxData :
STRUCT
	arrSeqData : ARRAY[0..1] OF INT;
	bActicated : BOOL;
END_STRUCT
END_TYPE
```

And my PLC.MAIN:

```
PROGRAM MAIN
VAR
	stLinuxData : ST_LinuxData;
	iLinuxClk	: UINT;
	bStart : BOOL;		
END_VAR
```

Everything works fine as long as I have TwinCAT open on a laptop, however, as soon as I open the TwinCAT on PLC or connect an HMI as a new ads client, I get 'Port: 30000 already in use as 0x7ffe9a7eba50' error.