stlehmann/pyads #425: Better distribution of `pyads` through platformed wheels

RobertoRoos, 2024-09-19:
I was thinking about for a little bit and wanted to open issue this issue mostly to host some discussion.

Currently `pyads` is published only as a source package, no wheels at all: https://pypi.org/project/pyads/#files.
When installed on Linux/Mac, `adslib` is always compiled during install and the resulting `.so` library is dumped amidst the `pyads` source code, where it's dynamically loaded at runtime.

This definitely works, but I feel it's not according to modern standard and could be improved.

I would suggest we distribute mainly as wheels instead. So we'd publish:

- A wheel for Linux, already including the compiled `adslib` (and no `adslib` source)
- A wheel for MacOS, already including the compiled `adslib` (and no `adslib` source)
- A wheel for Windows, containing only the `pyads` source and no `adslib` source or binaries
- A source distribution for platforms falling through the cracks, with the `adslib` source and no binaries so they can be compiled during installation

This way most installations shouldn't require any compilation at all.  
I'm pretty far with an example of this already.

CagtayFabry, 2026-01-14:
Thank you for your continous work on the project build process @RobertoRoos @stlehmann , especially with ` 3.5.1` 🚀 

With the new version out, are we any closer to build or providing wheels? That would make distribution much easier for a few setups

RobertoRoos, 2026-01-14:
I kind of lost track of the status here. I think I did finish this completely iin #426 , and it got rebased and merged in #469 

And on PyPi I see that the release from this week (`3.5.1`) actually includes a wheel for the first time (for Linux): https://pypi.org/project/pyads/3.5.1/#files (unlike for example: https://pypi.org/project/pyads/3.4.2/#files)  
Downloading and extracting the wheels shows a `adslib.so`, so I think it's all done?

@CagtayFabry are you in a position to test if this new wheel works on those setups, without compilation on the spot?

closed 2026-01-14 by RobertoRoos