stlehmann/pyads #432: add_route_to_plc fails under Linux

vrees, 2024-11-20:
I want to connect to my Beckhoff PLC, CX5110, running TwinCAT 3.1, build 4024.62 via ADS protocol under Linux. While trying to **add_route_to_plc** an error occurred in the pyads lib:

[image]

`onnected to pydev debugger (build 242.23339.11)
Traceback (most recent call last):
  File "/home/vrees/.local/share/JetBrains/IntelliJIdea2024.2/python-ce/helpers/pydev/pydevd.py", line 1570, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vrees/.local/share/JetBrains/IntelliJIdea2024.2/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/vrees/work/git-repos/heatpump2influxdb/main.py", line 15, in <module>
    pyads.add_route_to_plc(CLIENT_NETID, CLIENT_IP, PLC_IP, PLC_USERNAME, PLC_PASSWORD,route_name=ROUTE_NAME)
  File "/home/vrees/work/git-repos/heatpump2influxdb/venv/lib/python3.11/site-packages/pyads/ads.py", line 200, in add_route_to_plc
    return adsAddRouteToPLC(
           ^^^^^^^^^^^^^^^^^
  File "/home/vrees/work/git-repos/heatpump2influxdb/venv/lib/python3.11/site-packages/pyads/pyads_ex.py", line 169, in wrapper
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/vrees/work/git-repos/heatpump2influxdb/venv/lib/python3.11/site-packages/pyads/pyads_ex.py", line 375, in adsAddRouteToPLC
    raise RuntimeError(f"Unexpected response from PLC: {data!r}")
RuntimeError: Unexpected response from PLC: b"\x03f\x14q\x00\x00\x00\x00\x06\x00\x00\x80\x05|\xca\xe0\x01\x01\x10'\x01\x00\x00\x00\x01\x00\x04\x00\x04\x07\x00\x00"
python-BaseException`

The debug output shows, that the PLC answers with reasonable data. But the check if the received password is correct, fails as the expected data does not match the true or false condition. It looks like that the parsed data is somehow wrong interpreted.
I double checked username and password as it is still the default password.

[image]

chrisbeardy, 2024-11-21:
hmmmm not seen this error before! It looks like the plc is sending a repsonse that we are not expecting, because the password should either be correct or not. I wonder if Beckhoff have changed the expected response, I've not tested myself with 4024.62, thats quite a new one. You don't have the ability to try another runtime?

vrees, 2024-11-21:
The CX5110 uses a Windows CE7 Operating System. Changing the Runtime is quite complex as it uses SD cards for the installation. And I was happy to get it running so far. I would be very reluctant to change that.
Any other ideas?

ToBeNumberOneOne, 2025-03-03:
[image] Any thing update? i met the same issue with ubuntu22 and twicat 4024.55

ToBeNumberOneOne, 2025-03-03:
Seems like udp raw data get a wrong message