stlehmann/pyads #439: Pyads not working on MAC osx with M1 processor

rudy93, 2024-12-04:
Hi all, 

Got someone Pyads working on a Macbook with M1 processor? I receive the following error after installing Pyads:

    import pyads
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyads/__init__.py", line 10, in <module>
    from .ads import (
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyads/ads.py", line 50, in <module>
    from .pyads_ex import (
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pyads/pyads_ex.py", line 91, in <module>
    _adsDLL = ctypes.CDLL(adslib)
              ^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: dlopen(adslib.so, 0x0006): tried: 'adslib.so' (no such file), '/System/Volumes/Preboot/Cryptexes/OSadslib.so' (no such file), '/usr/lib/adslib.so' (no such file, not in dyld cache), 'adslib.so' (no such file), '/usr/lib/adslib.so' (no such file, not in dyld cache)

chrisbeardy, 2024-12-05:
Currently, the Apple M series is not supported, unfortunately. This is being worked on, however this project is only worked on during the spare time of a few participants so we do not have a timeline for this feature.

rudy93, 2024-12-05:
Okay thank you for the info, no problem I will use Linux instead. Do you guys know if I will be able to use Pyads on my M1 if I run a Linux virtual machine?

chrisbeardy, 2024-12-05:
Not sure, give it a go.

mrpasztoradam, 2026-09-22:
> Should be solved with the introduction of wheels for Mac OS [#498](https://github.com/stlehmann/pyads/pull/498). Please check out by installing from test.pypi:
> 
> ```
> pip install --index-url=https://test.pypi.org/simple/ pyads-stlehmann
> ```

Verified on Apple Silicon (M-series, macOS, arm64, Python 3.14): pip install --index-url=https://test.pypi.org/simple/ pyads-stlehmann installs the macosx_11_0_arm64 wheel and import pyads works — the bundled adslib.so is a native arm64 Mach-O. Port opens, a write/read round-trip against pyads.testserver works, and connecting to a real TwinCAT PLC works too (state, device info, symbol upload and typed reads all fine).